PX4/PX4-Autopilot #12725: Compile Firmware Errors "about cmake and build/"

qinglixinlei, 2019-08-17:
make  px4_fmu-v2_default
make[1]: 正在进入目录 `/home/PX4/workspace/PX4/FIR/Firmware'
-- PX4 config file: /home/PX4/workspace/PX4/FIR/Firmware/boards/px4/fmu-v2/default.cmake
CMake Error at platforms/nuttx/cmake/px4_impl_os.cmake:123 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "prebuild_targets".

Call Stack (most recent call first):
  cmake/px4_add_board.cmake:227 (px4_os_prebuild_targets)
  boards/px4/fmu-v2/default.cmake:2 (px4_add_board)
  CMakeLists.txt:215 (include)

-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- PX4 version: v1.9.0-649-g7adc43b
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 4.9.3
-- The C compiler identification is GNU 4.9.3
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-21-g62fa464
CMake Error at src/lib/version/CMakeLists.txt:69 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "git_ver".

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:252 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:331 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:358 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:359 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:360 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:361 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:362 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:363 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:364 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:365 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:353 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:366 (add_nuttx_dir)

-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- Configuring incomplete, errors occurred!
See also "/home/PX4/workspace/PX4/FIR/Firmware/build/px4_fmu-v2_default/CMakeFiles/CMakeOutput.log".
See also "/home/PX4/workspace/PX4/FIR/Firmware/build/px4_fmu-v2_default/CMakeFiles/CMakeError.log".
Error: /home/PX4/workspace/PX4/FIR/Firmware/build/px4_fmu-v2_default is not a directory

qinglixinlei, 2019-08-17:
I have encountered this error ,while first compiling firmware.some one can help me？

julianoes, 2019-08-19:
I think your cmake version is too low. I suggest to use Ubuntu 16.04 to build.;