PX4/PX4-Autopilot #12734: Creating a Fast RTPS Listener application: build error

dedyc, 2019-08-18:
**Describe the bug**
Trying to follow the instructions in [https://dev.px4.io/v1.9.0/en/middleware/micrortps.html]
FastRTPS is installed from sources (Tried several versions).
When I build the micrortps_agent code I get the following errors:

[  1%] Building CXX object CMakeFiles/micrortps_agent.dir/RtpsTopics.cpp.o
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:63,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_.h:67:15: error: redefinition of ‘const uint8_t position_setpoint__SETPOINT_TYPE_POSITION’
 const uint8_t position_setpoint__SETPOINT_TYPE_POSITION = 0;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:62,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_.h:66:15: note: ‘const uint8_t position_setpoint__SETPOINT_TYPE_POSITION’ previously defined here
 const uint8_t position_setpoint__SETPOINT_TYPE_POSITION = 0;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:63,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_.h:68:15: error: redefinition of ‘const uint8_t position_setpoint__SETPOINT_TYPE_VELOCITY’
 const uint8_t position_setpoint__SETPOINT_TYPE_VELOCITY = 1;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:62,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_.h:67:15: note: ‘const uint8_t position_setpoint__SETPOINT_TYPE_VELOCITY’ previously defined here
 const uint8_t position_setpoint__SETPOINT_TYPE_VELOCITY = 1;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:63,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_.h:69:15: error: redefinition of ‘const uint8_t position_setpoint__SETPOINT_TYPE_LOITER’
 const uint8_t position_setpoint__SETPOINT_TYPE_LOITER = 2;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:62,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_.h:68:15: note: ‘const uint8_t position_setpoint__SETPOINT_TYPE_LOITER’ previously defined here
 const uint8_t position_setpoint__SETPOINT_TYPE_LOITER = 2;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:63,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_.h:70:15: error: redefinition of ‘const uint8_t position_setpoint__SETPOINT_TYPE_TAKEOFF’
 const uint8_t position_setpoint__SETPOINT_TYPE_TAKEOFF = 3;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:62,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_.h:69:15: note: ‘const uint8_t position_setpoint__SETPOINT_TYPE_TAKEOFF’ previously defined here
 const uint8_t position_setpoint__SETPOINT_TYPE_TAKEOFF = 3;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_.h:75:15: note: ‘const uint8_t position_setpoint__VELOCITY_FRAME_BODY_NED’ previously defined here
autouser@ubuntu:~/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/build$

TSC21, 2019-08-18:
@dedyc can you please clone build and install Fast-RTPS 1.8.1 or above? This was a bug reported and correct already on the upstream Fast-RTPS repository.

TSC21, 2019-08-18:
Related issue: https://github.com/eProsima/Fast-RTPS/issues/534

dedyc, 2019-08-18:
Tried but the problem persists.
Currently installed versions:
PX4:          1.9.2
Fast-CDR: 1.0.10
Fast-RTPS: 1.9.0

Fast-CDR:

Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/share/fastcdr/LICENSE
-- Up-to-date: /usr/local/include/fastcdr
-- Up-to-date: /usr/local/include/fastcdr/eProsima_auto_link.h
-- Installing: /usr/local/include/fastcdr/FastBuffer.h
-- Up-to-date: /usr/local/include/fastcdr/exceptions
-- Up-to-date: /usr/local/include/fastcdr/exceptions/NotEnoughMemoryException.h
-- Up-to-date: /usr/local/include/fastcdr/exceptions/BadParamException.h
-- Up-to-date: /usr/local/include/fastcdr/exceptions/Exception.h
-- Up-to-date: /usr/local/include/fastcdr/Cdr.h
-- Up-to-date: /usr/local/include/fastcdr/FastCdr.h
-- Up-to-date: /usr/local/include/fastcdr/fastcdr_dll.h
-- Installing: /usr/local/include/fastcdr/config.h
-- Installing: /usr/local/lib/libfastcdr.so.1.0.10
-- Installing: /usr/local/lib/libfastcdr.so.1
-- Up-to-date: /usr/local/lib/libfastcdr.so
-- Installing: /usr/local/share/fastcdr/cmake/fastcdr-targets.cmake
-- Installing: /usr/local/share/fastcdr/cmake/fastcdr-targets-release.cmake
-- Installing: /usr/local/share/fastcdr/cmake/fastcdr-config.cmake
-- Installing: /usr/local/share/fastcdr/cmake/fastcdr-config-version.cmake

Fast-RTPS:
[100%] Built target fastrtps
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/share/fastrtps/LICENSE
-- Up-to-date: /usr/local/include/fastrtps
-- Installing: /usr/local/include/fastrtps/Domain.h
-- Up-to-date: /usr/local/include/fastrtps/fastrtps_fwd.h
-- Up-to-date: /usr/local/include/fastrtps/common
-- Installing: /usr/local/include/fastrtps/common/KeyedChanges.h
-- Up-to-date: /usr/local/include/fastrtps/fastrtps_dll.h
-- Up-to-date: /usr/local/include/fastrtps/eProsima_auto_link.h
-- Up-to-date: /usr/local/include/fastrtps/qos
-- Installing: /usr/local/include/fastrtps/qos/LivelinessLostStatus.h
-- Installing: /usr/local/include/fastrtps/qos/WriterQos.h
-- Installing: /usr/local/include/fastrtps/qos/DeadlineMissedStatus.h
-- Installing: /usr/local/include/fastrtps/qos/ReaderQos.h
-- Installing: /usr/local/include/fastrtps/qos/LivelinessChangedStatus.h
-- Installing: /usr/local/include/fastrtps/qos/ParameterTypes.h
-- Installing: /usr/local/include/fastrtps/qos/QosPolicies.h
-- Installing: /usr/local/include/fastrtps/qos/ParameterList.h
-- Installing: /usr/local/include/fastrtps/TopicDataType.h
-- Up-to-date: /usr/local/include/fastrtps/participant
-- Up-to-date: /usr/local/include/fastrtps/participant/ParticipantListener.h
-- Installing: /usr/local/include/fastrtps/participant/Participant.h
-- Up-to-date: /usr/local/include/fastrtps/log
-- Up-to-date: /usr/local/include/fastrtps/log/StdoutConsumer.h
-- Installing: /usr/local/include/fastrtps/log/Log.h
-- Up-to-date: /usr/local/include/fastrtps/log/Colors.h
-- Up-to-date: /usr/local/include/fastrtps/log/FileConsumer.h
-- Up-to-date: /usr/local/include/fastrtps/attributes
-- Up-to-date: /usr/local/include/fastrtps/attributes/all_attributes.h
-- Installing: /usr/local/include/fastrtps/attributes/SubscriberAttributes.h
-- Installing: /usr/local/include/fastrtps/attributes/PublisherAttributes.h
-- Installing: /usr/local/include/fastrtps/attributes/ParticipantAttributes.h
-- Installing: /usr/local/include/fastrtps/attributes/TopicAttributes.h
-- Up-to-date: /usr/local/include/fastrtps/types
-- Installing: /usr/local/include/fastrtps/types/DynamicTypeMember.h
-- Installing: /usr/local/include/fastrtps/types/TypeIdentifierTypes.h
-- Installing: /usr/local/include/fastrtps/types/DynamicTypeBuilder.h
[... 667 lines cut ...]
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:63,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_.h:75:15: error: redefinition of ‘const uint8_t position_setpoint__VELOCITY_FRAME_LOCAL_NED’
 const uint8_t position_setpoint__VELOCITY_FRAME_LOCAL_NED = 1;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_PubSubTypes.h:29:0,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_Subscriber.h:47,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:62,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_.h:74:15: note: ‘const uint8_t position_setpoint__VELOCITY_FRAME_LOCAL_NED’ previously defined here
 const uint8_t position_setpoint__VELOCITY_FRAME_LOCAL_NED = 1;
[... cut ...]
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:62,
                 from /home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:33:
/home/autouser/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_seaautautouauaaaaaaaaautouser@ubuntu:~/git/PX4/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/build$

julianoes, 2019-08-19:
@dedyc please put ``` around copy pasted console output to make it more readable.

closed 2019-11-17 by TSC21