PX4/PX4-Autopilot #12748: How to send position in offboard mode for VTOL with fixed wing mode?

hana9090, 2019-08-20:
How to send position in offboard mode for VTOL with fixed-wing mode?
I am sending a position using this topic: 

`ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped> ("mavros/setpoint_position/local", 10);`

But it behaves as multi-rotor not as fixed wing while moving. 
What is the right way to set position for fixed-wing in offboard mode?

Jaeyoung-Lim, 2019-08-20:
@hana9090 The reason it moves as if it a multirotor is becuase you are running the vehicle in multirotor mode.

In Fixedwing mode, position setpoints are not supported. The support will be added with https://github.com/PX4/Firmware/pull/12532, but has not been tested with a VTOL.

hana9090, 2019-08-20:
So this pull request is still not merged to px4 master repository, isn't it?

Jaeyoung-Lim, 2019-08-20:
@hana9090 Yes, as you can see in the link

Jaeyoung-Lim, 2019-08-21:
@hana9090 It is now merged

closed 2019-10-10 by Jaeyoung-Lim