PX4/PX4-Autopilot #12757: POSITION_TARGET_LOCAL_NED stream not appearing

v01d, 2019-08-20:
**Describe the bug**
Quadcopter set to altitude mode (on ground, disarmed) with PX4FLOW sensor attached. No POSITION_TARGET_LOCAL_NED stream on QGC visible

**To Reproduce**
Steps to reproduce the behavior:
1. Quadcopter with PX4FLOW attached, disarmed
2. Altitude mode set via radio transmitter (confirmed in QGC)
3. No POSITION_TARGET_LOCAL_NED stream visible in QGC analyze

**Expected behavior**
I expect to be able to switch to altitude mode and observe the setpoint in order to tune the altitude controller. Same thing happens if set to position control with local position input from onboard computer. I'm using FMUv2 board. I tried observing the visible mavlink streams from TELEM1 (set to "Normal" as per default) and from USB port. PX4 version is 1.9.2.

**Log Files and Screenshots**
No log as the drone was not armed.

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Custom built quadcopter
- Pixhawk FMUv2 with PX4FLOW sensor attached via I2C

julianoes, 2019-08-21:
I think you would need to request the MAVLink message stream using:
https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL

v01d, 2019-08-21:
According to the code I've read, this stream is supposed to be on for "normal" and "onboard" stream types.

julianoes, 2019-08-21:
Ok, and what mode are you using?

v01d, 2019-08-21:
This is on default TELEM1 port with Normal mode (the default).

julianoes, 2019-08-22:
I'm trying to reproduce this and I can see `POSITION_TARGET_LOCAL_NED` being streamed at around 0.5 Hz when flying in HITL in MISSION, POSCTL, and ALTCTL.

However, the values in ALTCTL are all `NaN` except `vz`.

[image]

In POSCTL everything seems sane, except `z` is still `NaN`. I wonder why that happens, presumably it's just not passed to this message.

[image]

julianoes, 2019-08-22:
@bresch it looks like the z setpoint is `NaN` for ALTCTL and POSCTL. I think that's not intuitive and makes it weird to tune.

julianoes, 2019-08-22:
The issue where the z setpoint is NaN is a recent problem -> #12776.

v01d, 2019-08-22:
Maybe it is because of this that the mavlink message is not visible (some not Nan guard avoid publication)?

julianoes, 2019-08-22:
> Maybe it is because of this that the mavlink message is not visible (some not Nan guard avoid publication)?

Let me check!

julianoes, 2019-08-22:
This works for me now with latest master (the screenshot is SITL) but this should be the same thing over telem1 just only 0.5 Hz.

[image]

v01d, 2019-08-22:
I've flashed current master on a standalone Pixhawk board (just RC receiver connected) and, while I can switch to altitude mode, I still don't get position target stream (neither on USB port nor TELEM1 port):

[image]

julianoes, 2019-08-23:
Ok, and that's with a GPS lock?

v01d, 2019-08-23:
No, no lock. This is a board with only RC attached. It is on altitude mode using baro.

julianoes, 2019-08-26:
This might be the issue then. Can you check if you see the message with GPS?

julianoes, 2019-08-26:
> Do you mean that the setpoint will not be visible without GPS? In altitude mode I believe it should appear even without GPS since it is sending an altitude setpoint to follow with barometer. Right?

I agree that it should but first I need to understand why it doesn't.

PR #12780 (2019-08-22): SmoothVel - Fix altitude lock logic
It was broken because _velocity_setpoint is used for input and output
and was assumed to be the input at a place where it was already overwitten
To clarify this, the input setpoint is renamed "target"

fixes #12776  and fixes #12757