PX4/PX4-Autopilot #12759: ESC calibration not calibrating low end

andrwp, 2019-08-20:
**Describe the bug**
The ESC calibration routine seems to output PWM_MAX for a number of seconds after plugging in the battery, then drops to the disarmed value. This results in a calibration abort in the ESC since the disarmed value is too low.

**To Reproduce**
Perform an ESC calibration while monitoring the PWM signal or listening to the ESC beep codes.

**Expected behavior**
1. ESC calibration should start with PWM_MAX for a number of seconds, as it currently does.
2. ESC calibration should progress to output PWM_MIN for a number of seconds to calibrate the low throttle value.

**Log Files and Screenshots**
I can capture the PWM values during calibration with an oscilloscope if that helps.

**Drone (please complete the following information):**
I'm using a Holybro Pixhawk 4 with stock PX4 1.9.2 and T-Motor Flame 60 HV ESCs.

julianoes, 2019-08-21:
Can you check how low the PWM min is that you're seeing? Generally this used to work, so I'm wondering if this is specific to your hardware.

I think you can also do `pwm info` in nsh to get that information.

dusan19, 2019-08-21:
I discovered this too, in March, and proposed this solution back then.
https://github.com/PX4/Firmware/pull/11663

dusan19, 2019-08-21:
For me the calibration works, it just sets the min PWM setting in the ESC to the minimum value instead to the one that is defined by the MIN_PWM PX4 parameter.
Note: I used a blheli_s ESC to detect the problem and test the solution.

julianoes, 2019-08-21:
Thanks for pitching in @dusan19.
So just to clarify, I believe that the calibration should be from PWM_DISARMED to PWM_MAX. That's because PWM_MIN is a setting where the motors should be spinning at an idle throttle.

dusan19, 2019-08-21:
If that is the case, that is what is going on at the moment. 
But in this case, the ESC sets its min PWM setting to the minimal allowed value (if disarmed is default 900), and we are sending the control signals from MIN_PWM to MAX_PWM, which means that we have a dead zone from the ESC's min setting to the PX4's MIN_PWM, which further means that the mapping between the desired control signals, and the control signals that the ESC decodes from the PWM is not straightforward.
For example if the MIN_PWM setting was not low, but say 1400, and max was 2000, that would mean that if we wanted 50% throttle we would send 1700 ms PWM pulse which the esc would threat as ~70% signal or so, since it maps from the disarmed to max value.

julianoes, 2019-08-21:
> For example if the MIN_PWM setting was not low, but say 1400, and max was 2000, that would mean that if we wanted 50% throttle we would send 1700 ms PWM pulse which the esc would threat as ~70% signal or so, since it maps from the disarmed to max value.

But the ESCs don't have a notion of 50% throttle it's all some relative scalar as far as I know.

andrwp, 2019-08-21:
> Can you check how low the PWM min is that you're seeing? Generally this used to work, so I'm wondering if this is specific to your hardware.

The PWM_DISARMED value that is being sent after the PWM_MAX segment of calibration is 899us pulse width (measured with an oscilloscope and a logic analyzer), which is too low of a value for my T-Motor Flame ESCs to recognize. I had to set PWM_MIN to 1140us to get my motors to spin after attempting calibration using PX4 1.9.2. I then manually calibrated my ESCs to 2000us max and 1000us min and was able to set PWM_MIN to 1050 to get the motors to spin. That seemed a lot more reasonable and verified that PX4 calibration was not working since you would expect my new PWM_MIN value to be higher with the manual calibration value that I used.

It seems to me that we should be calibrating to 2000us and 1000us and then restricting the output during armed operation to PWM_MIN and PWM_MAX. I don't think we should be attempting to calibrate ESCs to a value outside of the commonly accepted 1000us-2000us range, but it's probably ok to send a lower value when disarmed to ensure the motors won't possibly spin in that case.

> But the ESCs don't have a notion of 50% throttle it's all some relative scalar as far as I know.
I think that's true as long as the ESCs aren't using a closed loop mode like RPM control.

julianoes, 2019-08-22:
> I don't think we should be attempting to calibrate ESCs to a value outside of the commonly accepted 1000us-2000us range

Completely agree.

> I think that's true as long as the ESCs aren't using a closed loop mode like RPM control.

I think even then it's all relative. At least the controller is not assuming any relationship like that as far as I know.

ndepal, 2019-08-22:
> I think even then it's all relative. At least the controller is not assuming any relationship like that as far as I know.

Relative to what?

The ESC just sees a (pseudo) PWM signal at its input and converts that to a duty cycle on the motor voltage (for ESCs without RPM control, e.g. BLHeli ESCs). So the ESC somehow decides what percentage of the full battery voltage to apply. The only way it can do that is by determining what percentage is being asked of it, no?

julianoes, 2019-08-22:
Right but 50% doesn't have any physical meaning except 50% between no rotation and max RPM of that ESC/motor setup if RPM control is available. RPM doesn't translate to thrust in a linear way anyway. And completely stopping is not something you want because the control response has a huge delay, so the range is capped anyway.

I suppose it would be nice if 50% PWM aka 1500us would map to 50% RPM and then the controlled would be operating from e.g. 0.1 to 1 which would be 1100us to 2000us. I'm not sure if that's the case.

Maybe @bresch you can help me here, I argued myself into a corner :smile:.

ndepal, 2019-08-22:
What it comes down to is that the ESC has a notion of "50% input" from the FC. That means the ESC has to map a given PWM signal (e.g. 1500us) into some range of min (0%) and max (100%). So, if the FC and ESC do not have the same understanding of what those are, the FC can't demand the full range of motor outputs.

In the example that @dusan19 made, if the FC outputs 1400 for idle thrust, but the ESC is expecting values between 1000 and 2000 (because the calibration was not done right), it will make the motor spin much faster than what you might want, it might even be above hover thrust depending on your setup.

julianoes, 2019-08-22:
> In the example that @dusan19 made, if the FC outputs 1400 for idle thrust, but the ESC is expecting values between 1000 and 2000 (because the calibration was not done right)

Right, so our calibration should be something like 1000 to 2000 and disarmed should be 1000, and the operating PWM (MIN to MAX) something like 1100..1200 to 2000.

MaEtUgR, 2023-08-07:
Reading these reports was useful while addressing the issue for PX4 1.14. I had some problems myself and really wanted to solve the problems one by one. For 1.14 we now have:
- Improved PWM ESC calibration always from 1000 to 2000us https://github.com/PX4/PX4-Autopilot/pull/21711
- Conservative motor PWM configuration defaults and the possibility to intentionally stop e.g. tailsitter motors in fixed wing with any output protocol: https://github.com/PX4/PX4-Autopilot/pull/21513
- Documentation revised for the adjusted PWM calibration and configuration: https://docs.px4.io/main/en/advanced_config/esc_calibration.html

Please let me know if there are use cases that might still not be straightforward and reference the reports here. Then we can further improve the user experience.

closed 2023-08-07 by MaEtUgR