PX4/PX4-Autopilot #12761: GF_ACTION "Hold" fails to prevent geofence violation while in Hold-reposition

superware, 2019-08-21:
**Describe the bug**
When GF_ACTION is "Hold" and there's a geofence violation while the drone is in Hold-reposition (QGroundControl click on map and "Go to location"), the drone continues to the commanded location which is outside the geofence.

**To Reproduce**
Steps to reproduce the behavior:
1. Set GF_ACTION to Hold (Loiter).
2. Upload a geofence to the drone.
3. Takeoff, click somewhere outside the geofence (or inside if it's a keep-out fence) and select "Go to location".
4. Observe the drone violating the geofence and continuing to the destination position.

**Expected behavior**
The drone must reposition to the closest location on the geofence border, even if it's already in Hold/Loiter mode! Simply switching to Hold/Loiter isn't enough.

bozkurthan, 2019-08-21:
Hello @superware ,
Actually there was no problem with GF_ACTION when I tested with real flight. 

Can you check GF_COUNT parameter to sure about problem is related to PX4 stack.

> GF_COUNT
> 
>  Geofence counter limitComment: Set how many subsequent position measurements outside of the fence are needed before geofence violation is triggered
>

superware, 2019-08-21:
Hello @bozkurthan ,

> Actually there was no problem with GF_ACTION when I tested with real flight.

Did you try to violate the geofence while already in Hold (repositioning to another location)? This is easily reproducible with SITL.

> Can you check GF_COUNT parameter to sure about problem is related to PX4 stack.

GF_COUNT: -1, I am using PX4.

julianoes, 2019-08-21:
Thanks for the issue. This sounds like the implementation is broken for this case.

bozkurthan, 2019-08-21:
@superware 
Nope, you are right I just tried with manual mode.
I reproduced this behavior.

GF_ACTION
[image]

Fence limit
[image]

Violation by using "Go to Location"
[image]

BazookaJoe1900, 2020-01-11:
its still relevant

hamishwillee, 2020-04-14:
This needs to be fixed :-(

BazookaJoe1900, 2020-05-28:
related to: 'MC Geofence violation not easy to resolve #12320'

sfuhrer, 2023-11-15:
This has been fixed in the meantime (double actually: there is a check that doesn't allow you to set a goto outside of the fence, and upon hitting the fence the vehicle would now stop there).

closed 2023-11-15 by sfuhrer