PX4/PX4-Autopilot #12767: Review termination and lockdown failsafe

julianoes, 2019-08-21:
We need to review the handling of termination and lockdown flags as it came up in the discussion in https://github.com/PX4/Firmware/pull/12709.

We also need to decide if lockdown is actually a safety "feature" or just a setting used for HITL.

FYI @jlecoeur

jlecoeur, 2019-08-21:
Related, termination is sometimes associated with failsafe, sometimes not. See https://github.com/PX4/Firmware/pull/12763

jlecoeur, 2019-11-19:
Bad bot

julianoes, 2019-11-20:
> Bad bot

Just ignore him :smirk:.

cesnovar, 2021-05-17:
A somewhat related question to this stale issue: Is there a place flight termination is reported?

I was thinking it was reported in the `system_status` of the `Heartbeat` Mavlink stream, but it seems that triggering flight termination (using the VEHICLE_CMD_DO_FLIGHTTERMINATION command) does not put the UAV in the `vehicle_status_s::NAVIGATION_STATE_TERMINATION` state. 

Is this intentional behavior? If so, is there another way to check whether the UAV is in flight termination?