PX4/PX4-Autopilot #12768: Not able to connect or flash Pixhawk Cube

julianoes, 2019-08-21:
There have been reports of Pixhawk Cube's not connecting to QGC and not detected for flashing.

https://discuss.px4.io/t/unable-to-connect-with-qgc-with-pixhawk-cube-2-1/12520

Presumably on start no LEDs are on except one orange LED solid on.

Brought up by @RomanBapst.
FYI @davids5

julianoes, 2019-08-22:
It shows up as:
```
[291148.647785] usb 1-2: new full-speed USB device number 45 using xhci_hcd
[291148.776486] usb 1-2: New USB device found, idVendor=2dae, idProduct=1001, bcdDevice= 2.00
[291148.776492] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[291148.776496] usb 1-2: Product: CubeBlack-BL
[291148.776498] usb 1-2: Manufacturer: Hex/ProfiCNC
[291148.776501] usb 1-2: SerialNumber: 370036000851383236333932
[291148.778892] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```

And it shows up as:
`/dev/ttyACM0`
and:
`/dev/serial/by-id/usb-Hex_ProfiCNC_CubeBlack-BL_370036000851383236333932-if00`

It looks like we don't match for that in our tooling yet, fixed in: #12768.

LorenzMeier, 2019-08-22:
The link to the fix is probably not correct.

julianoes, 2019-08-22:
@LorenzMeier thanks for getting me out of that loop :smile:.

PR #12772 (2019-08-22): platforms: match usb-Hex_ProfiCNC device
This means that we are able to detect and flash the Pixhawk Cube Black.

We probably need to do the same in QGC.

Fixes #12768.
closed 2019-08-22 by davids5