PX4/PX4-Autopilot #12773: FastRTPS installation broken

julianoes, 2019-08-22:
It looks like the FastRTPS installation is broken.

I'm following https://dev.px4.io/master/en/setup/fast-rtps-installation.html and I'm stuck at:

```
cmake -DTHIRDPARTY=ON -DBUILD_JAVA=ON ..
-- Setting build type to 'Release' as none was specified.
-- The C compiler identification is GNU 9.1.1
-- The CXX compiler identification is GNU 9.1.1
-- Check for working C compiler: /usr/lib64/ccache/cc
-- Check for working C compiler: /usr/lib64/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/lib64/ccache/c++
-- Check for working CXX compiler: /usr/lib64/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring Fast RTPS
-- Version: 1.9.0
-- To change the version modify the file configure.ac
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Success
-- Performing Test SUPPORTS_CXX1Y
-- Performing Test SUPPORTS_CXX1Y - Success
-- Performing Test SUPPORTS_CXX11
-- Performing Test SUPPORTS_CXX11 - Success
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- fastcdr thirdparty is being updated...
Submodule 'thirdparty/fastcdr' (https://github.com/eProsima/Fast-CDR.git) registered for path 'thirdparty/fastcdr'
Cloning into '/home/julianoes/src/Fast-RTPS/thirdparty/fastcdr'...
Submodule path 'thirdparty/fastcdr': checked out '0f148fcdcbdd019214ef32632f035544c9645378'
-- Configuring Fast CDR
-- Version: 1.0.10
-- To change the version modify the file configure.ac
-- fastcdr library found...
Submodule 'thirdparty/asio' (https://github.com/chriskohlhoff/asio.git) registered for path 'thirdparty/asio'
Cloning into '/home/julianoes/src/Fast-RTPS/thirdparty/asio'...
Submodule path 'thirdparty/asio': checked out '22afb86087a77037cd296d27134756c9b0d2cb75'
Submodule 'thirdparty/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'thirdparty/tinyxml2'
Cloning into '/home/julianoes/src/Fast-RTPS/thirdparty/tinyxml2'...
Submodule path 'thirdparty/tinyxml2': checked out '8c8293ba8969a46947606a93ff0cb5a083aab47a'
-- Found tinyxml2: /home/julianoes/src/Fast-RTPS/thirdparty/tinyxml2  
CMake Error at CMakeLists.txt:215 (find_package):
  By not providing "Findfoonathan_memory.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "foonathan_memory", but CMake did not find one.

  Could not find a package configuration file provided by "foonathan_memory"
  with any of the following names:

    foonathan_memoryConfig.cmake
    foonathan_memory-config.cmake

  Add the installation prefix of "foonathan_memory" to CMAKE_PREFIX_PATH or
  set "foonathan_memory_DIR" to a directory containing one of the above
  files.  If "foonathan_memory" provides a separate development package or
  SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/julianoes/src/Fast-RTPS/build/CMakeFiles/CMakeOutput.log".
```

FYI @TSC21.

I looked into this because an issue was raised here:
https://discuss.px4.io/t/error-with-make-px4-sitl-rtps/12563/4

julianoes, 2019-08-22:
It looks like Fast-RTPS now needs this dependency, instructions how to build and install it are in
https://github.com/eProsima/Fast-RTPS#manual-installation, however, they don't work since it only installs it locally and not in the system.

And there are some discussion and workarounds here:
https://github.com/eProsima/Fast-RTPS/issues/620

Why is this so messly?

TSC21, 2019-08-22:
@julianoes we are currently only supporting Fast-RTPS 1.8.1. Also, I am not sure about the error above as I never faced it before. What system are you running this on?

TSC21, 2019-08-22:
The above issue seems only to be introduced in 1.9.0 and master. I advise you to clone the 1.8.x branch or install it from the binaries.
```sh
wget -q "http://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-7-2/eprosima_fastrtps-1-7-2-linux-tar-gz?format=raw" -O /tmp/eprosima_fastrtps.tar.gz \
&& cd /tmp && tar zxf eprosima_fastrtps.tar.gz \
&& cd eProsima_FastRTPS-1.7.2-Linux \
&& ./configure CXXFLAGS="-g -D__DEBUG" --libdir=/usr/lib \
&& make install \
&& rm -rf /tmp/*
```

julianoes, 2019-08-22:
Ok fine. This means we should update the instructions in the devguide.

TSC21, 2019-08-22:
> Ok fine. This means we should update the instructions in the devguide.

I agree. I have been really busy with the development that I dropped the docs for a while. I will add that to my TODO list with high priority. Thanks for reporting @julianoes.

julianoes, 2019-08-22:
Docs update: https://github.com/PX4/Devguide/pull/857

julianoes, 2019-08-22:
Ok, so v1.8.1 builds for me for the cpp part but not the java part:
```
:buildIDLParser
:idl:clean
:idl:generateGrammarSourceerror(7):  cannot find or open file: ../antlr4/omg/com/eprosima/idl/parser/grammar/IDL.g4
 FAILED
:buildIDLParser FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':idl:generateGrammarSource'.
> There was 1 error during grammar generation

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org
```

TSC21, 2019-08-22:
Damn they still haven't updated the release binaries --'

Related issue: https://github.com/eProsima/Fast-RTPS/issues/641

You will have to clone the `1.8.x` branch, build it and install it.

julianoes, 2019-08-22:
> You will have to clone the 1.8.x branch, build it and install it.

That's what I tried.

TSC21, 2019-08-22:
> > You will have to clone the 1.8.x branch, build it and install it.
> 
> That's what I tried.

Are you sure that you have the submodules synced?

TSC21, 2019-08-22:
Just delete the build folder, make sure that you have the submodules synced with that branch and rebuild it.

julianoes, 2019-08-22:
> Are you sure that you have the submodules synced?

Yes.

I tried after I wiped the build folder, same thing.

TSC21, 2019-08-22:
> > Are you sure that you have the submodules synced?
> 
> Yes.
> 
> I tried after I wiped the build folder, same thing.

Now that's awkward. The fix should be in that branch already. I am not sure why are you bumping to that error.

TSC21, 2019-08-22:
This is the commit that fixes the issue: https://github.com/eProsima/Fast-RTPS/commit/cfaf5307dae3ab94e6ff929f7507779585895691

richiware, 2019-08-27:
FYI. eProsima/Fast-RTPS#675 updates the README to recommend the installation of _foonathan_memory_ using our vendor package.

```bash
    $ git clone https://github.com/eProsima/foonathan_memory_vendor.git
    $ cd foonathan_memory_vendor
    $ mkdir build && cd build
    $ cmake ..
    $ cmake --build . --target install
```

Our vendor package solves the problem about the installation directory of _foonathan_memory_ library.

TSC21, 2019-08-27:
> How does the Fast-RTPS build find the foonathan_memory_vendor install since it's installed in tree and not in `/usr/local/` or `/usr`?

@julianoes I would say you need to use `sudo make install`.

TSC21, 2019-08-27:
> @TSC21 actually @richiware was right, it looks fixed now, so the `--target install` already tries to install it to `/usr/local`.

Yeah makes sense.

julianoes, 2019-08-27:
How does the Fast-RTPS build find the foonathan_memory_vendor install since it's installed in tree and not in `/usr/local/` or `/usr`?

julianoes, 2019-08-27:
@TSC21 actually @richiware was right, it looks fixed now, so the `--target install` already tries to install it to `/usr/local`.

closed 2019-09-26 by TSC21

prafull199905, 2025-03-18:
-- OpenSSL library 3.3.1 found...
-- Not using a GNU compiler on Windows.
CMake Error at src/cpp/CMakeLists.txt:9 (find_package):
  By not providing "Findfastrtps.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "fastrtps",
  but CMake did not find one.

  Could not find a package configuration file provided by "fastrtps" with any
  of the following names:

    fastrtpsConfig.cmake
    fastrtps-config.cmake

  Add the installation prefix of "fastrtps" to CMAKE_PREFIX_PATH or set
  "fastrtps_DIR" to a directory containing one of the above files.  If
  "fastrtps" provides a separate development package or SDK, be sure it has
  been installed.

 i face this issue many times why ? even i installed many times Fast-DDS and Fast-CDR 
please guide me