PX4/PX4-Autopilot #12779: VTOL: run fixed-wing land detector if in FW mode

sfuhrer, 2019-08-22:
Currently, only the multicopter land detector is run on VTOL vehicles, and the landed state is set to "not landed" if in fixed-wing. As though for many VTOLs also a purely fixed-wing landing would be possible, it would be nice it would run the fixed-wing land detector in fixed-wing mode. 

With https://github.com/PX4/Firmware/pull/12777 I've made a fix that sets the landed state to "landed" only if disarmed.

dlwalter, 2019-09-06:
This is great, and solves a particular problem where your transition switch/threshold isn't set on a new VTOL airframe and the aircraft is stuck in FW mode.  You can't get into the Flight Modes or Radio settings to fix it because you are locked out of those settings.

sfuhrer, 2019-09-11:
> This is great, and solves a particular problem where your transition switch/threshold isn't set on a new VTOL airframe and the aircraft is stuck in FW mode. You can't get into the Flight Modes or Radio settings to fix it because you are locked out of those settings.

Yeah this really sucked..but this is actually already fixed with https://github.com/PX4/Firmware/pull/12777. This issue here is to allow landing being detected (while arming) in FW mode.