PX4/PX4-Autopilot #12789: Abnormal hover in RTL mode

jiejiezhang, 2019-08-23:
Description of the problem: Frequent switching mode (RTL<------->POSITION), there is a probability that an abnormal hover will occur in rtl mode. The position_setpoint_triplet.lat and position_setpoint_triplet.lon are not published is at this time.

So I am a bit confused about here(https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L723-L736): Why do we publish only once after updating the position setpoint ?

julianoes, 2019-08-23:
Thanks for the issue @jiejiezhang. Do you have a log to show what happens?

jiejiezhang, 2019-08-24:
[image]

So I guess it is because the position setpoint here((https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L723-L736) is only sent once after each updated.

julianoes, 2019-08-26:
@MaEtUgR @bresch FYI

bresch, 2022-08-15:
I've not see this recently, please re-open if it still occurs

closed 2022-08-15 by bresch