PX4/PX4-Autopilot #12791: Snapdragon not working (again)

julianoes, 2019-08-23:
It looks like PX4 on Snapdragon is yet again broken.

The good new is it does build, link and run, however:
- The DSP side startup script px4.config doesn't seem to be run. I need to run all commands using `qshell some_module start`.
- `uorb top` doesn't seem to work on the apps side but the `listener` does work.
- `sensor_gyro` is published but `sensor_combined` is not even though `sensors` is running.

If anyone is currently running Snapdragon and has hints or fixes, please share them, thanks.

julianoes, 2019-09-01:
@dagar I don't see the error above.

I got further by using this as mainapp.config:
```
#!/bin/sh

# PX4 commands need the 'px4-' prefix in bash.
# (px4-alias.sh is expected to be in the PATH)
. px4-alias.sh

uorb start
muorb start
sleep 1

qshell param set SYS_AUTOSTART 4001
qshell sleep 1
qshell gps start -d /dev/tty-4
qshell param set MAV_TYPE 2
qshell sleep 1
#qshell df_hmc5883_wrapper start
#qshell df_mpu9250_wrapper start_without_mag
qshell df_mpu9250_wrapper start
qshell df_bmp280_wrapper start
qshell df_trone_wrapper start
qshell #df_ltc2946_wrapper start # (currently not working on all boards...)
qshell #df_isl29501_wrapper start
qshell sleep 1
qshell sensors start
qshell commander start
qshell ekf2 start
qshell land_detector start multicopter
qshell mc_pos_control start
qshell mc_att_control start
qshell snapdragon_pwm_out start
qshell spektrum_rc start

logger start -t -b 200
# Wait 1s before setting parameters for muorb to initialize
sleep 1
param set MAV_BROADCAST 1
dataman start
navigator start
mavlink start -x -u 14556 -r 1000000
sleep 1
mavlink stream -u 14556 -s HIGHRES_IMU -r 50
mavlink stream -u 14556 -s ATTITUDE -r 50
mavlink stream -u 14556 -s RC_CHANNELS -r 20
mavlink boot_complete
```

and then running `./px4 -s mainapp.config`.

In order to subscribe to topics on the appsproc side I need to do `listener <topicname>` twice, then it starts flowing and e.g. appears over MAVLink.

dagar, 2019-09-01:
@julianoes push to `pr-snapdragon_hacks` for testing on hardware. https://github.com/PX4/Firmware/commits/pr-snapdragon_hacks

bresch, 2022-08-15:
Please re-open if you plan to fix it

closed 2022-08-15 by bresch