PX4/PX4-Autopilot #12804: wind estimator

moreba1, 2019-08-25:
Hi
we need to see wind speed and direction in flight for safety reasons.

[image]

dagar, 2019-08-25:
Which version of PX4 are you using and how is it configured? Do you have a differential pressure sensor and the EKF2 airspeed fusion threshold set (EKF2_ARSP_THR)?

https://github.com/PX4/Firmware/blob/32fb4a958566c3855cab5b2e414b45d4f4dc9a65/ROMFS/px4fmu_common/init.d/rc.fw_defaults#L22-L31

moreba1, 2019-08-25:
i used latest version of px4 with vtol deltaquad airframe reference . I use an airspeed sensor and actived it. EKF2_ARSP_THR was set to 0 (default) . do i have to change it to stall speed ?

moreba1, 2019-08-26:
There is need documentation for wind estimation enabling .
does we can set maximum allowable wind speed for flight safety ?

RomanBapst, 2019-08-26:
@moreba1 Can you share the log of that flight?

moreba1, 2019-08-28:
https://review.px4.io/plot_app?log=b6146ff1-7a47-4259-a6ed-be9f05e1d172

sfuhrer, 2019-08-30:
@moreba1 there is no "maximum allowable wind speed" preflight check, because wind estimation only starts in flight. 
For wind estimation to be active you need to set  EKF2_ARSP_THR to about 10m/s for your vehicle. Problem in your case: your airspeed sensor gives about 30% too low readings when compared to the ground velocity, so there won ever be a good wind estimate out of it. (as you mentioned in your issue https://github.com/PX4/Firmware/issues/12676)

Can you share a picture of where your airspeed sensor pitot is mounted? Maybe there's a problem there.

closed 2019-08-30 by moreba1