PX4/PX4-Autopilot #12813: fmu-v4_default build fails

baumanta, 2019-08-27:
**Describe the bug**
When I try to build current master for v4, the build fails with the following error message.

../../src/lib/led/led.cpp: In member function 'void LedController::get_control_data(LedControlData&)':
../../src/lib/led/led.cpp:229:5: error: this statement may fall through [-Werror=implicit-fallthrough=]
     if (cur_data.blink_times_left % 10 < 6) { // 2 blinks, then turn off for the rest of the cycle
     ^~
compilation terminated due to -Wfatal-errors.

**To Reproduce**
Steps to reproduce the behavior:
1. checkout master@2f10c315b56f0bd9cb200956b112ad1f0930b90a
2. make px4_fmu-v4_default

**Additional context**
I use the gcc toolchain version:
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

julianoes, 2019-08-27:
The question is why you have a newer `arm-none-eabi` version installed than recommended in:

https://github.com/PX4/Devguide/blob/8456e743470793414b6eda119844b8a320c138c9/build_scripts/ubuntu_sim_nuttx.sh#L29-L44

or:
https://github.com/PX4/Firmware/blob/2f10c315b56f0bd9cb200956b112ad1f0930b90a/Tools/setup/ubuntu.sh#L115-L140

julianoes, 2019-08-27:
@baumanta can you do `which arm-none-eabi-gcc` please?

baumanta, 2019-08-27:
@julianoes that returns: /usr/lib/ccache/arm-none-eabi-gcc

julianoes, 2019-08-27:
Ok, and that's a symlink. Where does it go?

```
ls -l /usr/lib/ccache/
```

julianoes, 2019-08-28:
@hamishwillee so you're saying the build is broken in general, not just with @baumanta's newer compiler?

hamishwillee, 2019-08-28:
I get same error. Installed from ubuntu.sh. 

```
ubuntu@ubuntu-VirtualBox:~/github/Firmware$ which arm-none-eabi-gcc
/opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
```

julianoes, 2019-08-28:
Ok, this is really weird with the script from `ubuntu.sh` it fails to compile with the fallthrough error. With the scripts from the devguide it's fine.'

I tried to find differences and found that the arm-none-eabi-gcc version as well as the cmake versions are identical. The only difference is that the Firmware script also installs `ccache`.

I tried to remove the ccache cache using `ccache -C` but it did not resolve the issue! Only `sudo apt-get remove ccache` helped.

julianoes, 2019-08-28:
Presumably the comments are stripped by the preprocessor before it goes into ccache. The workaround `export CCACHE_CPP2=yes` seems to work for me.

hamishwillee, 2019-08-28:
I'm saying that if you install with ubuntu.sh from that PR we've been discussing, then the problem happens irrespective of the build target - ie FMUv2, 3, 4 etc.

baumanta, 2019-08-29:
@julianoes about the symlink, running your command, I get:
lrwxrwxrwx 1 root root 16 Nov 26  2018 arm-none-eabi-g++ -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 arm-none-eabi-gcc -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Mär 12 08:57 arm-none-eabi-gcc-7.3.1 -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 c++ -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 c89-gcc -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 c99-gcc -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 cc -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 g++ -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 g++-5 -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 gcc -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 gcc-5 -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 x86_64-linux-gnu-g++ -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 x86_64-linux-gnu-g++-5 -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 x86_64-linux-gnu-gcc -> ../../bin/ccache
lrwxrwxrwx 1 root root 16 Nov 26  2018 x86_64-linux-gnu-gcc-5 -> ../../bin/ccache

baumanta, 2019-08-29:
I just mentioned the build target, because I noticed that SITL builds are fine. I did not try any other targets, e.g. v2 or such

MaEtUgR, 2019-10-09:
I updated the title since we long found out what the problem and possible fix is:
https://github.com/PX4/Firmware/pull/12820#issuecomment-527415694
https://github.com/PX4/Firmware/pull/12827#issuecomment-527958040
To still work without manual ccache upgrade the compile warnings were temporarily disabled:
https://github.com/PX4/Firmware/pull/12825/files#diff-903c2a19ede406370227cbda7ee03051R40
https://github.com/PX4/Firmware/pull/12901/files#diff-dfaa0b66e8c5239fe60fe79fd5b29749R36
https://github.com/PX4/Firmware/commit/1f63d85869b3495f3f66a3300b365c57469f1020#diff-f56b373864d883619e4d0961090a9e9aR52 (not sure if NuttX implements the fallthrough guards)
And the desired solution is mentioned multiple times
https://github.com/PX4/Firmware/pull/13104#issuecomment-538816225
https://github.com/PX4/Firmware/pull/13084#issuecomment-537914922
Switching to C++17 and using the `[[fallthrough]];` statement.

Let's keep this issue to remember/track that. I'm not aware what's missing to be fully C++17 compatible. @dagar mentioned the compiler version for one board `once the snapdragon toolchain is updated`.

hamishwillee, 2020-01-12:
So is this still something that needs to be alive? ie is this just alive as a reminder to people there is a problem, or are we planning on doing something?

julianoes, 2020-02-12:
For Ubuntu 16.04 users this can be resolved by installing a newer ccache version as it is done in:

https://github.com/PX4/Firmware/blob/0d36e5094b4655c960846d481f62361d40cf1d9f/Tools/setup/ubuntu.sh#L103-L107

closed 2020-02-12 by julianoes