PX4/PX4-Autopilot #12814: NuttX build warning with arm-none-eabi-gcc 8-2018-q4 8.2.1 20181213

julianoes, 2019-08-27:
With the next arm-none-eabi-gcc version:
```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213
```

we run into the following warning and fail the build:

```
make px4_fmu-v4_default
[334/338] Generating nuttx/sched/libsched.a
FAILED: cd /home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx && find sched -type f -name *.o -delete && make -C sched --quiet --no-print-directory all TOPDIR=/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx KERNEL=y EXTRADEFINES=-D__KERNEL__ >nuttx_build.log
wdog/wd_start.c: In function 'wd_expiration':
wdog/wd_start.c:151:20: error: cast between incompatible function types from 'wdentry_t' {aka 'void (*)(int,  unsigned int,  ...)'} to 'void (*)(int)' [-Werror=cast-function-type]
                 (*((wdentry0_t)(wdog->func)))(0);
                    ^
cc1: all warnings being treated as errors
make[1]: *** [wd_start.o] Error 1
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'px4_fmu-v4_default' failed
```

This is related to:
https://github.com/PX4/NuttX/pull/53
https://github.com/PX4/NuttX/pull/52
https://bitbucket.org/nuttx/nuttx/issues/159/sched-wdog-fix-signature-for-wdentry_t
https://github.com/PX4/NuttX/blob/px4_firmware_nuttx-7.29%2B/configs/z16f2800100zcog/tools/zneo-zdsii-5_0_1-variadic-func-fix.patch

FYI @davids5

WangxingboWWW, 2019-08-31:
I have the same problem
wdog/wd_start.c: In function 'wd_expiration':
wdog/wd_start.c:151:20: error: cast between incompatible function types from 'wdentry_t' {aka 'void (*)(int,  unsigned int,  ...)'} to 'void (*)(int)' [-Werror=cast-function-type]
                 (*((wdentry0_t)(wdog->func)))(0);
And I used the newest arm-none-eabi-gcc version: gcc-arm-none-eabi-8-2019-q3-update
How can I fix this bug?

MaEtUgR, 2019-09-25:
That issue was about the exact same error: https://github.com/PX4/Firmware/issues/11410

julianoes, 2020-01-30:
@davids5 do you know if this is still the same or resolved in the meantime in upstream NuttX?

davids5, 2020-01-30:
@julianoes  - Yes I believe it is in upstream.

closed 2020-01-30 by davids5