PX4/PX4-Autopilot #12817: VTOL No Global Position Failsafe

moreba1, 2019-08-27:
Hi
I had a problem with No Global Position Failsafe failsafe.

https://review.px4.io/plot_app?log=672c5564-9512-42a8-adad-2aed5138d92e

julianoes, 2019-08-28:
Please give some more context on what you are trying to do and what happened.

julianoes, 2019-08-28:
The estimator complains about yaw and mag problems which might indicate a bad mag calibration or interference of motors/electronics.

Also, the jamming indicator of the GPS is quite high. Presumably other components like camera or companion computers are interfering.

moreba1, 2019-08-28:
> 
> 
> Please give some more context on what you are trying to do and what happened.

i planned a vtol take off and transition and rest mission. plane performed vertical takeoff but it was unstable in yaw and roll then it switched to fixed wing mode . a few moments later it gave a massage "No Global Position Failsafe" and automaticaly switched to stablized mode (dangerous work). 

What is this error reason ?
1- mag problem
2- GPS jamming problem
3- accelerometers problem because of vibration
4- airspeed sensor problem

julianoes, 2019-08-28:
@sfuhrer FYI

sfuhrer, 2019-08-30:
@moreba1 The fixed-wing motor current seems to interfere your mag a lot:
[image]

Also there is a tonn of noise, especially close before the No Position failsafe was triggered.

The combined accel and mag problems have probably triggered no position failure I assume. 

Did I get this correctly, you are using one of the internal (inside the Pixhawk) magnetometers and don't have an external one? This is usually not a good idea, as there usually is a lot of power lines close to the autopilot. 

Beside that, your vehicle is very badly tuned in hover by the looks of it from the rate plots. I'd increase the roll/pitch PID values, or start clean by following https://docs.px4.io/v1.9.0/en/config_mc/pid_tuning_guide_multicopter.html

moreba1, 2019-08-30:
@sfuhrer  Thank you .
can we disable ekf dependency to mag by ekf2_mag_type ? and In addition to reducing vibration can we reduce weight of accels and gyro in position estimation ?

sfuhrer, 2019-08-30:
@moreba1 accels alone don't give you good attitude estimates, it needs accel, gyro and mag for that. But yes, I guess you can disable mag fusion with EKF2_MAG_TYPE I guess. 

Concerning the fibrations: they're mainly bad for attitude/rates estimates, position comes from GPS. So if you can fly nicely in stabilized then it's fine I guess with the amount of noise you have there.

moreba1, 2019-08-30:
OK

closed 2019-08-30 by moreba1