PX4/PX4-Autopilot #12832: Startup script without errors, no Parameter XXX not found

BazookaJoe1900, 2019-08-28:
I Think that if during start up parameter is not found is shouldn't be an error, but info. because current implementation assume that its ok not to have parameter, and just treat its non-exist value as 0. 
This is relevant to comparison of parameter. Setting non-exist parameter should be kept as error
Unless you REALY need the parameter, and its mandatory. I don’t' know if there is such competition that REALY need the parameter. 
Do you agree?

julianoes, 2019-08-29:
I think I agree. Can you provide an example that you're seeing. And suggest where or how we can improve that?

BazookaJoe1900, 2019-08-30:
I have tested it again on pixhawk 4, (fmu-v5, master)  and I don't see "Parameter XXX not found" errors.
I guess I seen errors because I looked at costum firmware that I run, that removes drivers that are not relevant to me. but in fmu-v5 most of the driver are being compiled.
This is still potential, for example ``` if param compare SENS_EN_LL40LS 2```  or ```if param greater SENS_EN_MB12XX 0``` will trigger the error if the LL240LS or MB12XX driver is not compiled. because its in common rc script

on my configuration I see:
```
ERROR [param] Parameter SENS_EN_BATT not found
ERROR [param] Parameter SENS_EN_MB12XX not found
ERROR [param] Parameter SENS_EN_PGA460 not found
ERROR [param] Parameter SENS_EN_SF1XX not found
ERROR [param] Parameter SENS_EN_TRANGER not found
ERROR [param] Parameter TRIG_MODE not found
ERROR [param] Parameter UAVCAN_ENABLE not found
```

there are few options how to handle on-exist parameter
1) keep current behavior - if there is comparison to parameter that doesn't exists, trigger error. I don't like it because its trigger many false errors
2) Remove any Error - Don't like it again, because there are some cases that parameter must be exist, and if it doesn't, its really error. for example, many SYS_AUTOSTART comparison. if SYS_AUTOSTART doesn't exist (for any reason) its error
3) Reduce the error to Info - Don't like it too... its a middleway that just confuse and doesn't separate between real errors.
4) Create new "silent" comparation. that allows the parameter to be missed, and doesn't trigger any info/error. This is my prefer option

@julianoes 
What do you think?

There are other errors, and warnings on the startup, but not about parameters, but other staff.
```
[boot] Rev 0x0 : Ver 0x0 V500
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V5
HW type: V500
HW version: 0x00000000
HW revision: 0x00000000
FW git-hash: e50dd7c364350f02f03956941b3f88916072612e
FW version: 1.9.0 0 (17367040)
FW git-branch: master
OS: NuttX
OS version: Release 7.29.0 (119341311)
OS git-hash: d8da511082646d83a54c6905daca13f0a1a609f0
Build datetime: Aug 28 2019 05:42:56
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000200000000323634353237511800240029
MCU: STM32F76xxx, rev. Z
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
ERROR [safety_button] not starting (use px4io for safety button)
ERROR [safety_button] Task start failed (-1)
INFO  [px4_work_queue] creating: wq:lp_default, priority: 205, stack: 1700 bytes
INFO  [px4_work_queue] creating: wq:I2C1, priority: 248, stack: 1250 bytes
WARN  [rgbled] no RGB led on bus #1
WARN  [rgbled_ncp5623c] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
INFO  [px4_work_queue] creating: wq:hp_default, priority: 243, stack: 1500 bytes
Board sensors: /etc/init.d/rc.board_sensors
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 1 at 0 (1000 KHz)
INFO  [px4_work_queue] creating: wq:SPI1, priority: 254, stack: 1400 bytes
WARN  [mpu6000] unexpected ID 0xfd accepted
BMI055_ACCEL on SPI bus 1 at 3 (10000 KHz)
BMI055_GYRO on SPI bus 1 at 2 (10000 KHz)
INFO  [ist8310] no device on bus 1
INFO  [px4_work_queue] creating: wq:I2C2, priority: 247, stack: 1250 bytes
INFO  [ist8310] no device on bus 2
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [hmc5883] no device on bus 2 (type: 2)
WARN  [hmc5883] no device on bus 4 (type: 2)
WARN  [qmc5883] no device on bus 1 (type: 2)
WARN  [qmc5883] no device on bus 2 (type: 2)
WARN  [qmc5883] no device on bus 4 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
INFO  [px4_work_queue] creating: wq:I2C3, priority: 246, stack: 1250 bytes
IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
INFO  [px4_work_queue] creating: wq:SPI5, priority: 250, stack: 1400 bytes
ERROR [pmw3901] driver start failed
MS5611_SPI on SPI bus 4 at 0 (20000 KHz)
INFO  [px4_work_queue] creating: wq:SPI4, priority: 251, stack: 1400 bytes
INFO  [px4_work_queue] creating: wq:att_pos_ctrl, priority: 244, stack: 2000 bytes
INFO  [px4_work_queue] creating: wq:rate_ctrl, priority: 255, stack: 1600 bytes
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
```

dagar, 2019-09-01:
@BazookaJoe1900 blinkm might be another good one to put behind a parameter.

[image]

dagar, 2019-09-01:
FYI px4_work_queue spam will be silenced after https://github.com/PX4/Firmware/pull/12853.

[image]

dagar, 2019-09-01:
Lights (rgbled, rgbled_ncp5623c, rgbled_pwm) could be moved to board startup.

BazookaJoe1900, 2019-09-02:
for blinkm, pr: add new parameter LIGHT_EN_BLINKM to start blinkm lights #12880

BazookaJoe1900, 2019-09-02:
about pca8574, most, if not all boards compile it, but there is no 'pca8574 anywhere'.... What to do there?
I think of two options:
1)  is it possible that this driver is not relevant at all, so why to compile it at all
2) its relevan't but users that do use it, run it at run time by mavlink-shell, extra rcs.extra

BazookaJoe1900, 2019-09-02:
about the onboard leds, before moving them to the boards rcs, I want to cleanup and remove boards that might compile drivers that they don't have.
Here is list I found regaring rgbleds:
where there is no mark, the driver doesn't compiled
where you see '?' its mean I can't (and didn't) test if the HW exist and run well
Where you see 'ok' its mean that the CI [link](http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/detail/pr-sensors_orb_publish/3/pipeline/418 ) the driver and its seems to run ok.
where there is 'error' the CI failed to run the driver.
I don't know if there is any option to run external rgbleds from that kind.

platform | board | rgbled (device rgbled0) | rgbled_ncp5623c (device rgbled1) | rgbled_pwm
-- | -- | -- | -- | --
aerotenna | ocpoc | ? |   |  
airmind | mindpx-v2 | ? |   |  
atlflight | cmake_hexagon |   |   |  
atlflight | eagle | ? |   |  
atlflight | excelsior | ? |   |  
auav | esc35-v1 |   |   |  
auav | x21 | ? |   |  
av | x-v1 |   |   |  
beaglebone | blue |   |   |  
bitcraze | crazyflie |   |   |  
emlid | navio2 |   |   |  
holybro | kakutef7 |   |   |  
intel | aerofc-v1 |   |   |  
mro | ctrl-zero-f7 | ? |   |  
nxp | fmuk66-v3 | error | ok | ?
omnibus | f4sd | ? |   |  
parrot | bebop |   |   |  
px4 | cannode-v1 |   |   |  
px4 | esc-v1 |   |   |  
px4 | fmu-v2 | ok |   |  
px4 | fmu-v3 | ok | error | ?
px4 | fmu-v4 | error | error |  
px4 | fmu-v4pro | ok | error | ?
px4 | fmu-v5 | error | error | ?
px4 | fmu-v5x | error | error |  
px4 | io-v2 |   |   |  
px4 | raspberrypi |   |   |  
px4 | sitl |   |   |  
thiemar | s2740vc-v1 |   |   |  
uvify | core |   | ? |

dagar, 2019-09-03:
> about pca8574 and oraled, most, if not all boards compile it, but there is no 'pca8574 start'/ 'oraled start' anywhere'.... What to do there?
> I think of two options:
> 
> 1. is it possible that this driver is not relevant at all, so why to compile it at all
> 2. its relevan't but users that do use it, run it at run time by mavlink-shell, extra rcs.extra

The pca8574 driver could probably dropped unless someone would like to volunteer to maintain and test it. It has some custom IOCTLs that would be nice to purge.

The oreoled driver is apparently for the solo and isn't properly integrated with PX4. I'm inclined to drop it as well unless someone would like to change that. It's clearly not being used and ends up being a minor distraction when making larger changes or even looking for good examples.

LorenzMeier, 2019-09-03:
Please drop both drivers. We need to focus our resources on what is being used!

BazookaJoe1900, 2019-09-04:
@dagar @LorenzMeier there the list of the other leds (the table I added)
Do you want me to move the driver to each board rc.board_sensors? or rc.board_default?
I can guess some of the '?' I put on the table that need to be removed from compilation of the boards. some I can't tell. I can just keep them....

BazookaJoe1900, 2019-09-14:
@dagar ping

BazookaJoe1900, 2019-11-27:
@dagar ping, or should I close this issue?

dagar, 2019-11-27:
What would you like to do here? Personally I'd rather work towards addressing the problem more fundamentally, rather than hiding something we find ugly or annoying.

1. Some build time mechanism to register or install a driver or modules startup code. If you don't have that driver in the build the little bit of init could won't even be present and won't make the error.
2. Board manifest so that the set of required things we know are physically present in a configuration can be started automatically.
3. Deciding what to do about optional external things like i2c magnetometers. If we want to keep probing them it should be more intelligent (scanning the bus and only trying valid addresses, etc). Alternatively I don't think it would be unreasonable to have to set a parameter per bus for what you want.

BazookaJoe1900, 2019-11-28:
Agree, closing this issue

closed 2019-11-28 by BazookaJoe1900