PX4/PX4-Autopilot #12833: Parameter tuning MPC_ALT_MODE

Seeelefant, 2019-08-28:
**Describe the bug**
A clear and concise description of the bug.
I was trying to bind the parameter to (using parameter tuning in QGroundcontrol) to my low-medium-high settings of a 3-way switch, but only got
[image]
low:
[image]
medium:
[image]
high:
[image]
Please advice.
Best 
--Seeelefant

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Low-0: Altitude following
Medium-1: Terrain following
High-2: Terrain hold
**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

julianoes, 2019-08-29:
Thanks for the issue. I have the feeling the parameter tuning using a know was never tested/used for params with int type (not float). Feel free to debug it in code, maybe in HITL simulation. I'm not sure when someone else would look into this.

It's an interesting idea though to use the tuning knob functionality in this way.

Seeelefant, 2019-09-01:
Hi Julian,
thanks for your response...I suspected something like that. However, thinking the other way around: Within ArduCopter there is a continuous paramter called RNGFND_GAIN, which essentially interpolates between the height measured by the baro sensor and the rangefinder. Wouldn´t such a paramter make sense in the PX4 environment?

julianoes, 2019-09-03:
I don't know where a parameter like that would fit into the control and ekf2 set-up.

I would say we should fix the actual bug that you found here.

closed 2022-08-15 by bresch