PX4/PX4-Autopilot #12845: Stopping/Slowing at each Waypoint

Dan-Moss, 2019-08-29:
The aircraft slows to 1.2ms at each waypoint, even in a straight line. 
Firmware Version 1.9.2 on Black Cube

[image]
[image]

Param Changes:
[image]
NAV_ACC_RAD = 15

I'd expect changing the Cruise and Cruise 90 would allow the aircraft to pass through the waypoints without slowing, similar to the acceptance radius but it seems to not be working?

mhkabir, 2019-08-29:
@bresch

bresch, 2019-08-30:
Hi @Dan-Moss , this has been improved post 1.9.2 and is fixed by  #12505 
You can try the Master development branch, feedback is welcomed!

Dan-Moss, 2019-08-30:
Thanks for replying @bresch , will build and test shortly (i guess there is no instant way to get px4_fmu-v3_default.px4) . I will close this issue for now then?

mhkabir, 2019-08-30:
You can use QGC to flash the Developer build, which is autobuilt from master.

Dan-Moss, 2019-08-31:
@mhkabir when reconnecting to update firmware i'm getting:
Found device: PX4 FMU V2
Error: Sync: Send Command: Invalid sync response: 0xfe 0x10
Upgrade cancelled

v3.5.4 Qground
Tried a variety of PC's, Laptops and USB Cables?

mhkabir, 2019-08-31:
@dagar any clues?

Dan-Moss, 2019-09-01:
using http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-compile/activity/ i was able to get the build and it behaves correctly visually (Yay :D ) but the log shows:
[image]

closed 2019-11-29 by Dan-Moss