PX4/PX4-Autopilot #12855: MPU driver doesn't recognize device well

BazookaJoe1900, 2019-08-31:
**Describe the bug**
Looking at the nsh of pixhawk4 boot I can see the next line:

> WARN  [mpu6000] unexpected ID 0xfd accepted

Looking at the code, there is check of valid known product ID. but it seems that its isn't valid.

https://github.com/PX4/Firmware/blob/78ef8aab2df33cd00e2c969bbff37ffd82bcf385/src/drivers/imu/mpu6000/MPU6000.cpp#L246-L278

Looking at the ICM20689, the MPU6000 datasheets I don't even see the register (MPUREG_PRODUCT_ID, 0x0c) documented. Who knows where the ID list came from?

Except for this warning I don't see any strange behavior, but if some of the pixhawk 4 has new device, that might behave different, its better to know, and remove the warning.

**To Reproduce**
Enable nsh (it can be done to one of the telemetry ports).
and look at the output

BazookaJoe1900, 2019-12-02:
Still relevant.

mhkabir, 2019-12-02:
@davids5?

davids5, 2019-12-02:
It was early IP shared with PX4. It is there to tell us there is a new version of the IC has arrived - it is a only a warning.

BazookaJoe1900, 2019-12-05:
I also tried to get the information abot the tester from TDK with no luck...
Can you be sure that they have no effect? And if so why do we test them at all?

BazookaJoe1900, 2020-03-18:
@davids5 , its look like dead end.
what do you say, maybe, at least to remove the warning? or say that 0xfd is unknown, but in the list?

BazookaJoe1900, 2020-06-17:
dead end

closed 2020-06-17 by BazookaJoe1900