PX4/PX4-Autopilot #12860: Change how altitude is set in ORBIT mode

Dmitrykrz, 2019-09-01:
Orbit is a fantastic mode for shooting video. It works really well with gps module installed! Great feature.

There is just one problem. The altitude is controlled by throttle stick, <50% down, >50% up, 50% keeps altitude. It is very hard to keep the stick at precisely 50%, so copter slowly spirals up or down instead of orbiting.
This mode is designed for shooting video of object standing in the orbit center. I expect better autonomy without the need to adjust the throttle to keep the copter at same altitude.
Lets say I want to set the orbit and perform a dance in the middle without worrying the drone spiraling down or loosing me out of camera frame. 
What I propose?
Add parameter Orbit-Altitude-Throttle-Deadzone. The default is 0. When 0 the orbit works exactly like before, so guys are not confused who are used to old orbit.
When this parameter is set to 10 (for example) the 10% deadzone added. So now when throttle is 40-60% the drone will keep the altitude constant. <40% down, >60% up. 
The test is done on 1.9.2 firmware, standard 4 rotor drone with pixracer.

Dmitrykrz, 2019-09-02:
[image]

Here one can see that throttle was above 50% but still the Z setpoint was negative.

Dmitrykrz, 2019-09-03:
Maybe it was my mistake after all.
Correct me if I am wrong here.  Orbit follows MPC_HOLD_DZ ? If yes, then I just needed to set MPC_HOLD_DZ  to higher value thus achieving exactly what I wanted. We don't need extra parameters if MPC_HOLD_DZ  does what I need.

hamishwillee, 2019-12-02:
@MaEtUgR Can this be closed?

bresch, 2022-08-15:
The Z stick uses `MPC_HOLD_DZ`, closing.

closed 2022-08-15 by bresch