PX4/PX4-Autopilot #12866: Inconsistent vehicle state after changing BARO_SENS_QNH

aburgm, 2019-09-02:
**Describe the bug**
Changing the BARO_SENS_QNH value while the vehicle is on the ground changes its home position even though the vehicle has physically not moved. This leads to a discrepency between the home position altitude (where the vehicle thinks it is located) and the GPS reported altitude. After rebooting the vehicle, the discrepency disappears, but the new BARO_SENS_QNH value is still set, suggesting it was in some sort of inconsistent state before.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on (or SITL started)
2. Change BARO_SENS_QNH by 10 units up or down
3. Observe (e.g. in QGC) that home position altitude and GPS altitude no longer coincide
4. Reboot vehicle and they will coincide again

**Expected behavior**
A behavior of changing BARO_SENS_QNH to value `x` while the vehicle is on the ground that is consistent with booting the vehicle with BARO_SENS_QNH set to `x`. Alternatively, document that changing BARO_SENS_QNH requires a reboot.

**Context**
 * Maybe somewhat related to #11848.

 * I came across this when trying to feed external vision data using mavlink VISION_POSITION_ESTIMATE messages. Unfortunately the [corresponding documentation](https://dev.px4.io/v1.9.0/en/ros/external_position_estimation.html#reference-frames) does not state where the origin of the coordinates sent in that mavlink message should be, but I assume it should be the same origin as the vehicle reports in LOCAL_POSITION_NED. However, when changing QNH without rebooting the vehicle, my vision position coordinates are always differing in Z by a constant offset, because changing QNH changes the vehicle's (local) position.

dagar, 2019-09-02:
Just to clarify the parameter is `SENS_BARO_QNH`.

Yes I suppose that makes sense because changing the QNH will shift the barometer altitude (altimeter) output, but the estimator isn't aware. As a quick workaround we can mark this parameter with reboot required, but ideally we'd handle it online.

dagar, 2019-09-02:
https://github.com/PX4/Firmware/pull/12878

closed 2019-09-02 by dagar