PX4/PX4-Autopilot #12867: Airspeed setpoint is different with FW_AIRSPD_TRIM

moreba1, 2019-09-02:
Hi 

i noticed in flight test that airspeed setpoint in climb and loiter is different from FW_AIRSPD_TRIM but in my opinion , airspeed must be constant overal flight .

[image]

sfuhrer, 2019-09-02:
@moreba1 were you flying in mission mode? Probably would be best to share a log.

moreba1, 2019-09-02:
yes , flying is in mission mode.
https://review.px4.io/plot_app?log=b6146ff1-7a47-4259-a6ed-be9f05e1d172

sfuhrer, 2019-09-03:
@moreba1  The changes in airspeed setpoint happen when underspeed is detected (measured airspeed below FW_AIRSPD_MIN).

moreba1, 2019-09-03:
Why should the airspeed setpoint be increased if the airspeed is low? Throttle must increase to airspeed reach FW_AIRSPD_TRIM.  in this case  by this mistake airspeed has been increased by more than FW_AIRSPD_TRIM.

sfuhrer, 2019-09-04:
> Why should the airspeed setpoint be increased if the airspeed is low? Throttle must increase to airspeed reach FW_AIRSPD_TRIM. in this case by this mistake airspeed has been increased by more than FW_AIRSPD_TRIM.

The airspeed setpoint is reduced, not increased, if an airspeed undershoot is detected. So before the undershoot the airspeed setpoint was at TRIM, during the undershoot at MIN. The TECS message doesn't correspond with our parameter because it's true airspeed, not indicated as the parameter I think. Not sure about that though.

closed 2020-11-16 by moreba1

TruAero, 2025-05-05:
I dug into the PX4 info and found that the FW_AIRSPD_TRIM actually sets the airspeed setpoints for the equivalent airspeed, not true airspeed. I believe the true airspeed setpoints are automatically adjusted for changes changes in density from airspeed temperature and compressibility. There is no information on the equivalent airspeed definition in PX4 that I could find though.