PX4/PX4-Autopilot #12877: VTOL thrust during backtransition

sfuhrer, 2019-09-02:
While performing a transition from FW to MC in airspeed less mode, the thrust of the MC should linearly ramp up, starting when the transition is started. Instead, there is some little peak of MC throttle, then about 1s nothing, then a jump to about 30%:

[image]

dlwalter, 2019-09-06:
sfuhrer - outstanding.  I was just running into this exact issue last week.

sfuhrer, 2019-12-19:
@dlwalter do you still experience this issue? As far as I can say (don't have a lot of airspeed less flights) we don't with recent code.

dlwalter, 2019-12-19:
I haven't seen it in actual flight - I only saw this on the bench and that was months ago.

sfuhrer, 2019-12-20:
> I haven't seen it in actual flight - I only saw this on the bench and that was months ago.

Okay, same for me actually (at least the plot above was from a bench setup). I think it was related to the land detector detecting landing (which is fair, given that the vehicle is on the ground^^). 

Closing this then.

closed 2019-12-20 by sfuhrer