PX4/PX4-Autopilot #12879: VTOL: Kill switch not working during aborted back transiton

sfuhrer, 2019-09-02:
Flipping the killswitch during an aborted front transition doesn't seem to work. 

To reproduce:
- flip transition switch (to FW)
- flip transition switch back (fo MC)
- flip kill switch

The MC motors are correctly switched off, the FW pusher isn't. It keeps spinning, even after disarm. Only switches off when moved again into FW mode and killed there. 

[image]

dagar, 2019-09-02:
Likely fixed by https://github.com/PX4/Firmware/pull/12838.

The problem is that FW actuator control outputs (VTOL) were only being published on FW actuator control updates (from fw_att_control). The goal was to minimize some of the redundant mixing effort, but it's obviously wrong in various edge cases (in retrospect).

sfuhrer, 2019-09-03:
Yes fixed with #12838. Thanks @dagar ! (https://github.com/PX4/Firmware/issues/12877 is still valid though)

closed 2019-09-03 by sfuhrer