PX4/PX4-Autopilot #12886: Can we use PID

bluecat1997, 2019-09-03:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
A clear and concise description of the problem, if any, this feature will solve. E.g. I'm always frustrated when ...

**Describe your preferred solution**
A clear and concise description of what you want to happen.

**Describe possible alternatives**
A clear and concise description of alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots for the feature request here.

Antiheavy, 2019-09-06:
+1 for this feature request!

bresch, 2020-03-18:
To me, it still makes sense to add an optional D term for the pitch axis

bresch, 2020-06-25:
Anyone interested of implementing it?

RomanBapst, 2020-06-25:
@bresch @jlecoeur @sfuhrer I think what we should do is implemented this in the context of a unified rate controller for fixed wings and mulirotors.
We already have a separate work item for rate control, we just need to add feedforward and airspeed scaling. That is also very much aligned with future plans for control allocation.

jlecoeur, 2020-06-25:
@RomanBapst I agree

bresch, 2020-06-26:
@RomanBapst Awesome, I'm totally in favor of an unified controller.

closed 2022-08-19 by sfuhrer