PX4/PX4-Autopilot #12897: Param init undefined behaviour

julianoes, 2019-09-04:
On startup we get a warning by the clang undefined behaviour sanitizier:

This can be reproduced with:
https://github.com/PX4/Firmware/pull/12896

```
/home/julianoes/src/Firmwaresrc/lib/parameters/parameters.cpp:292:23: runtime error: null pointer passed as argument 2, which is declared to never be null
/usr/include/stdlib.h:819:6: note: nonnull attribute specified here
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /home/julianoes/src/Firmware/src/lib/parameters/parameters.cpp:292:23 in 
```

julianoes, 2019-09-04:
This fixes the warning but breaks the `param` command for SITL:

```
diff --git a/src/lib/parameters/parameters.cpp b/src/lib/parameters/parameters.cpp
index 3c5cad2879..b171a39d49 100644
--- a/src/lib/parameters/parameters.cpp
+++ b/src/lib/parameters/parameters.cpp
@@ -700,6 +700,10 @@ param_set_internal(param_t param, const void *val, bool mark_saved, bool notify_
                goto out;
        }
 
+       if (param_values->d == nullptr) {
+               goto out;
+       }
+
        if (handle_in_range(param)) {
 
                param_wbuf_s *s = param_find_changed(param);

```

junwoo091400, 2022-11-22:
🤔