PX4/PX4-Autopilot #12900: mavlink - reading to undefined memory

BazookaJoe1900, 2019-09-04:
First, I am not sure if this bug should be here or in mavlink.... I write it here, but it can moved.
I am not 100% sure about if my assumption is right.

**Describe the bug**
I played with @julianoes pr (posix: add fuzz testing using MAVLink messages #12896) and think that i might come to critical bug in mavlink_helper.h

when the fuzz sends msgid that is higher then the maximum message id on the mavlink_message_crcs[mid].msgid table. the mid parameter can go to the sizeof(mavlink_message_crcs) and there is a read from undefined address 
I think that the next statement is an error:
> high=sizeof(mavlink_message_crcs)/sizeof(mavlink_message_crcs[0])

it should be 

> high=sizeof(mavlink_message_crcs)/sizeof(mavlink_message_crcs[0])-1

this is table to show how low, high and mid iteration looks when msgid is higher then highest id on the table: assuming the number of messages is 186

low | high | mid
-- | -- | --
0 | 186 | 93
93 | 186 | 140
140 | 186 | 163
163 | 186 | 175
175 | 186 | 181
181 | 186 | 184
184 | 186 | 185
185 | 186 | 186 -> Error, calling to mavlink_message_crcs[186].msgid

**To Reproduce**
run posix: add fuzz testing using MAVLink messages #12896

julianoes, 2019-09-04:
Thanks @BazookaJoe1900, I forgot to add that you need to patch this. I usually did this locally.

See: https://github.com/ArduPilot/pymavlink/pull/343/files.

BazookaJoe1900, 2019-09-17:
@julianoes 
The fix has finelly merged on Mavlink. Closing this issue

closed 2019-09-17 by BazookaJoe1900

julianoes, 2019-09-18:
Thanks @BazookaJoe1900, however we still need to update the c_library_v2 submodule.

BazookaJoe1900, 2019-09-19:
@julianoes I assume that will be automatically.
do you want me to re-open?

zhangteng0526, 2024-05-07:
Is there any crash poc?