PX4/PX4-Autopilot #12902: severe wobble during landing RTL

mberendsen, 2019-09-05:
**Describe the bug**
I was doing my final flight test for my new drone build. I was testing transitioning between different flight modes using the RC transmitter and Ground Control on android. First I set several waypoints in a zig-zag pattern that came back to home. (30 meters alt & default speed) When the drone was passing the last wave point I used ground control app to direct the drone using the "go to here" feature when clicking a location on the map. 

The Drone moved to the selected location and then used the "go to here" two more times. On the third use of "go to here" function the Drone made a large swinging movement and it looked like the drone was going to flip. The drone corrected the the issue and flew normally again. The drone looked like it was going to land where it was and I accidentally hit the "RTL" switch.

The Drone moved to home without issue but it started wobbling severely while landing. When it got about 4-5 feet from the ground the wobbling started. It eventually flip and crashed at about 2-3 feet off the ground.   

Afterwards I recalibrated the compasses because I got the "px4 compass sensors inconsistent" error. I tried to recalibrate several times without success so I disabled the onboard compass. Now the low alt wobbling happens during any takeoff.  

**To Reproduce**
Steps to reproduce the behavior:
1. Arm Drone 
2. Put in stabilize  mode 
3. Take off 
4. Drone is steady for a couple seconds
5. Then At hover 2-4 feet off ground the drone will begin to wobble and crash.

**Expected behavior**
Fly low altitude without issue 

https://review.px4.io/plot_app?log=6431208f-5a4f-4f35-96bd-c2182f644cba

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
650mm tarot frame, 700kv 3111N tmotor, pixhawk 4, holybro PDB,
[image]

 40 amp ESC, 1555 tarot props with camera + gimbal, Lidar distance sensor for altitude estimate 

**Additional context**
Add any other context about the problem here.

julianoes, 2019-09-05:
Thanks for the detailed issue @mberendsen and sorry about the crash.

Could you elaborate about what you mean with wobble? And can you see this wobble in the log?

bresch, 2022-08-15:
High yaw demand with low thrust (descending) were probably causing this. Please re-open if you still have issues on newer versions of PX4

closed 2022-08-15 by bresch