PX4/PX4-Autopilot #12904: Motor Output Saturation

Dan-Moss, 2019-09-05:
**The bug**
Occasionally the motors seem to hunt on arming, meaning they pulse to max ouptut - Thus resulting in being unflyable. This bug does not happen all the time and see bellow a bad flight (left) compared to a good flight (right). These are both run on the same settings/params ect. To note is the amplitude of the controls and cutoff of the actuator outputs.

**Log Files and Screenshots**
[image]
[image]

[image]

**Drone**
Large Octoquad x8 running a Cube Black on PX4

**Additional context**
Attempted on Firmware; 1.9Dev, 1.9.2, 1.8.2 with same results

julianoes, 2019-09-25:
First, thanks for the issue. Now I'm confused what exactly I should see in the plots. What do you mean with "hunt"?

Dan-Moss, 2019-09-25:
@julianoes hunting - engine RPM fluctuating rapidly
the plots should be the same, but in the 'bad' occurrence it is maxing the pwm values supplied to the motors witch is also fluctuating rapidly.

julianoes, 2019-09-25:
Ok. Could you share a link to a log?

Dan-Moss, 2019-09-27:
https://logs.px4.io/plot_app?log=1c58e461-83b0-46d4-a11d-20e4c2ccc8f1

here is a link to the flight with the issue. Again this was an intermittent issue.

julianoes, 2019-10-01:
Thanks @Dan-Moss. I don't see what is causing this. My best guess is oscillations in the yaw controllers but I don't have data to back that up, and I don't see that you have increased yaw gains either.
It's also odd that it doesn't always happen.

@bresch do you see anything?

julianoes, 2020-01-30:
@Dan-Moss were you able to figure this one out? Or do you have additional hints?

Dan-Moss, 2020-02-03:
Hi @julianoes,
Could not figure it out, the workaround it just trying to get it when behaving as expected. Luckily can see the issue when arming / adding small throttle so no risk.

closed 2022-08-15 by bresch