PX4/PX4-Autopilot #12910: VTOL: no control in FW after transition in ALT control

sfuhrer, 2019-09-05:
reproducable in sitl: take off in alt, switch to FW. Not possible to change flight direction.

sitl log:
https://review.px4.io/plot_app?log=a60bffd8-f43f-4ca1-b59e-18d7dbe46bc9

flight log:
https://review.px4.io/plot_app?log=a14d1d71-bb8c-4438-a0bb-6ae8daed4190

sfuhrer, 2019-09-05:
fyi @dagar @RomanBapst

sfuhrer, 2019-09-05:
Throttle goes to 0 after transition to FW is complete (even though throttle stick is at 50%)
[image]

dlwalter, 2019-09-09:
I run into a similar issue but only when I don't have a good GPS lock.  Fixed wing/altitude I cannot get throttle, roll, or pitch to respond unless I have X number of satellites.

sfuhrer, 2019-09-10:
should be fixed when https://github.com/PX4/Firmware/pull/12920 is merged

closed 2019-09-10 by dagar