PX4/PX4-Autopilot #12911: Px4 Flow Sensor - Sensor data issues.

Ken-Thorburn, 2019-09-05:
I am trying to use a px4 flow sensor for position hold mode on a drone. Currently, I have tried using Ekf2 and LPE.

For Ekf2, the optical flow sensor is enabled and range is set as the height sensor, however, the flight controller will not allow me to enter into position hold mode. The drone is only receiving height information and not x and y (Seen in /mavros/local_position/pose). 

Additionally, I have tried using LPE, but the flight controller will only enter position hold mode with the parameters seen in the image one. This issue was also found in https://github.com/PX4/Firmware/issues/7600 in that the flight controller uses the camera for position, but does not use the sonar for height. 

Essentially, I have only been able to use the sonar for height with Ekf2 and the camera for position when using LPE.   

If anyone knows how to get this working, please let me know.

***Image One***
[image]

**Drone:**
- DJI F450
- Px4 Flow Sensor is connected via i2C bus shared with compass
- Photo of the IMU / autopilot setup
[image]

julianoes, 2019-09-25:
It would probably also help if you could share some logs for the cases that you see.

gouyiyong, 2019-10-29:
I have the same problem, /mavros/local_position/pose is not updating except z, but the optical flow data is published. I am using 1.9.0 version.

Ken-Thorburn, 2019-10-29:
@gouyiyong Have you tried flying and then switching into position hold mode?

gouyiyong, 2019-10-29:
> @gouyiyong Have you tried flying and then switching into position hold mode?

NO, just test no fly, x and y no any update, i dont know how to handle it, ekf2 issue?

gouyiyong, 2019-10-29:
NO, just test no fly, x and y no any update, i dont know how to handle it, ekf2 issue?

Ken-Thorburn, 2019-10-29:
I've tried bench testing it as well and have been unsuccessful. I will probably try flying it tomorrow, but my guess is that it prevents the x and y values of the px4 flow sensor from publishing to the pose until it has taken off and reached the minimum height and quality (default 1.5m and >150).

Ken-Thorburn, 2019-10-29:
@gouyiyong Are you using the sonar for height estimation or the barometer?

gouyiyong, 2019-10-29:
> @gouyiyong Are you using the sonar for height estimation or the barometer?

yes, i use lidar for height and px4flow for x and y

Ken-Thorburn, 2019-10-29:
What pixhawk fmu version are you using? @gouyiyong

gouyiyong, 2019-10-30:
> What pixhawk fmu version are you using? @gouyiyong
it is clone version

bresch, 2022-08-15:
This works on `main`

closed 2022-08-15 by bresch