PX4/PX4-Autopilot #12915: Rover stops while running mission (switch to RTL)

jarivanewijk, 2019-09-06:
As discussed on the call and on Slack yesterday, both @dk7xe and I encountered this issue.

When PX4 is put into mission mode, the rover will start driving towards its first waypoint but will stop almost immediately, before reaching the first point. In the logs we can see that it goes into RTL mode (but it just stops). Additionally, I have seen that the rover steers completely left or right (seemingly at random points) when it doesn't stop.

I am using a small Turnigy toy car with Ackermann steering to which I added a FMUK66. I am using the generic rover airframe. Gerald has a Aion R1 with a FMUK66.

The logs I posted on Slack:
> I am digging through my log files. Noticed this one where I tried to go into mission mode several times, but it almost immediately switches to RTL mode by itself. I think this is an example of the rover driving a small distance towards and waypoint and then stopping. https://logs.px4.io/plot_app?log=034e2db9-a33d-4249-b231-501a59aa470d (edited) 
> And I think in this one it went in the wrong direction: https://logs.px4.io/plot_app?log=de7bf18f-3e1f-4575-a2bf-5f3d0e5b9679
> This one as well: https://logs.px4.io/plot_app?log=0b9d9dfa-f49f-4d6d-b7a8-ea988224a955 (edited) 
> But I need to do further tests. These log files probably are not that useful.

And Gerald's:
> same behaviour at my R1 at RTL - https://review.px4.io/plot_app?log=62c8b7f4-d350-4c72-a122-2bdeea0a8a20

If the weather allows for it I will do more tests next week (and in a more structured manner). I will also need to review some of the parameters.

@dagar @ItsTimmy @julianoes

julianoes, 2019-09-19:
> same behaviour at my R1 at RTL - https://review.px4.io/plot_app?log=62c8b7f4-d350-4c72-a122-2bdeea0a8a20

When I look at this log I can see that the params for the AION R1 rover are not correctly set.

I suggest to go through all params in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/airframes/50003_aion_robotics_r1_rover and manually set them.

julianoes, 2019-12-19:
@jarivanewijk is this still an issue or can it be closed?

jarivanewijk, 2019-12-19:
@julianoes I haven't checked myself, but Gerald ( @dk7xe ) has found the cause of his issue. 

For him the steering was inverted, but RC transmitter output was also inverted. So everything worked in manual, but the rover would go into random directions when put in mission mode. I think everything worked when he changed the configuration.

I assume my issue is similar, but I haven't been able to test it and I probably won't for a while. You can close this, I'll open a new issue if needed.

julianoes, 2019-12-19:
Thanks, will do.

closed 2019-12-19 by julianoes