PX4/PX4-Autopilot #12923: i use gcc8.3.1 and now have a type error when make px4_fmu-v2

Richard-LXA, 2019-09-08:
error: cast between incompatible function types from 'void (*)(int, char**)' to 'main_t' {aka 'int (*)(int, char**)'} [-Werror=cast-function-type]
        (main_t)&PX4IO::task_main_trampoline,
                        ^~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors

jaejin0me, 2019-09-08:
"-Wno-cast-function-type" will be useful.
I recommend this article. [About -Wextra and -Wcast-function-type](https://debarshiray.wordpress.com/2019/04/01/about-wextra-and-wcast-function-type/)

julianoes, 2019-09-25:
I think this one just needs fixing. @Richard-LXA would you mind creating a pull request with a fix?