PX4/PX4-Autopilot #12930: [ VTOL ] Offboard support and MAVROS transition command

LightningPORTO, 2019-09-09:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

Ive been developing an autonomous VTOL based on ROS as a main brain of the hole operation to perform Search and rescue missions. The goal would be to have an autonomous vtol to increase the energy comsumpsion eficiency

Clearly, I need to have " full " offboard support and one topic to have a transition command and maybe even an hybrid configuration where both the multicopter and the fixed wing motors are working.

Thank you for all your time

julianoes, 2019-09-25:
So VTOL transition using MAVROS is missing?

Can you send generic commands in MAVROS?
For instance, that's possible with MAVSDK:
https://github.com/mavlink/MAVSDK/blob/fe06d5bafd2fb2180b04c389d1773c56b7876c79/src/plugins/action/action_impl.cpp#L169-L256

Jaeyoung-Lim, 2019-09-25:
@LightningPORTO You should be able to do it through `~cmd/command` service.

closed 2019-09-26 by julianoes

Jaeyoung-Lim, 2019-09-29:
@LightningPORTO Here you go: https://github.com/mavlink/mavros/pull/1323