PX4/PX4-Autopilot #12931: [Feature] Method to prevent VTOL arming while in Fixed Wing Mode

dlwalter, 2019-09-09:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
A clear and concise description of the problem, if any, this feature will solve. E.g. I'm always frustrated when ...

**Describe your preferred solution**
A clear and concise description of what you want to happen.

**Describe possible alternatives**
A clear and concise description of alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots for the feature request here.

junwoo091400, 2022-11-22:
@sfuhrer any ideas? Or whether this is necessary / feasible?

sfuhrer, 2022-11-23:
As far as I understand this was solved with https://github.com/PX4/PX4-Autopilot/pull/13987

closed 2022-11-23 by sfuhrer