PX4/PX4-Autopilot #12932: Cannot boot 13012 E-Flight Convergence on FMUv3 Cube

dlwalter, 2019-09-09:
**Describe the bug**
Boot hangs at mixer loading on a Pixhawk Cube (FMUv3).  I can't get to the nsh prompt and the console seems to hang at mixer loading

**To Reproduce**
1. Current Master branch (47e668eb86)
2. Set SYS_AUTOCONFIG 13012

**Expected behavior**
I expect to reach a nsh console during boot

**Log Files and Screenshots**
```
FMUv2 ver 0xE : Rev 0 V30
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V3
HW type: V30
HW version: 0x0009000E
HW revision: 0x00000000
FW git-hash: 47e668eb86531130bcc22b29f82d62eaa57704a6
FW version: 1.10.0 80 (17432704)
FW git-branch: master
OS: NuttX
OS version: Release 7.29.0 (119341311)
OS git-hash: d8da511082646d83a54c6905daca13f0a1a609f0
Build datetime: Sep  9 2019 12:14:34
Build uri: localhost
Toolchain: GNU GCC, 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
PX4GUID: 000100000000363137343336511100410033
MCU: STM32F42x, rev. 3
nsh: mount: mount failed: No such device
nsh: mkfatfs: mkfatfs failed: No such device
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 16
Board defaults: /etc/init.d/rc.board_defaults
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
INFO  [px4_work_queue] creating: wq:lp_default, priority: 205, stack: 1700 bytes
INFO  [px4_work_queue] creating: wq:I2C1, priority: 248, stack: 1250 bytes
INFO  [px4_work_queue] creating:WARN  [rgbled] no RGB led on bus wq:I2C2, priority: 247, stack:  #2
1250 bytes
WARN  [rgbled_ncp5623c] no RGB led on bus #2
nsh: rgbled_pwm: command not found
INFO  [px4_work_queue] creating: wq:hp_default, priority: 243, stack: 1500 bytes
Board sensors: /etc/init.d/rc.board_sensors
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
INFO  [ist8310] no device on bus 1
INFO  [ist8310] no device on bus 5
QMC5883_I2C on I2C bus 1 at 0x0d (bus: 100 KHz, max: 400 KHz)
WARN  [hmc5883] no device on bus 2 (type: 1)
WARN  [mpu6000] no device on bus #3 (SPI1)
MS5611_SPI on SPI bus 4 at 3 (20000 KHz)
INFO  [px4_work_queue] creating: wq:SPI4, priority: 251, stack: 1400 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
INFO  [px4_work_queue] creating: wq:SPI1, priority: 254, stack: 1400 bytes
WARN  [mpu6000] no device on bus #5 (SPI4)
INFO  [mpu9250] Bus probed: 5
MPU9250 on SPI bus 4 at 4 (1000 KHz)
INFO  [icm20948] Bus probed: 2
WARN  [icm20948] no device on bus 2
ERROR [ak09916] driver start failed
ERROR [ms5611] bus option already started
L3GD20 on SPI bus 4 at 1 (11000 KHz)
LSM303D on SPI bus 4 at 2 (11000 KHz)
INFO  [mpu9250] Bus probed: 3
MPU9250 on SPI bus 1 at 4 (1000 KHz)
Airspeed on I2C bus 2 at 0x21 (bus: 100 KHz, max: 100 KHz)
INFO  [px4_work_queue] creating: wq:att_pos_ctrl, priority: 244, stack: 2000 bytes
INFO  [px4_work_queue] creating: wq:rate_ctrl, priority: 255, stack: 1600 bytes
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ERROR [mavlink] DM_KEY_MISSION_STATE lock failed
ERROR [mavlink] offboard mission init failed (-1)
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/vtol_convergence.main.mix on /dev/pwm_output0
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
```

julianoes, 2019-09-25:
@davids5 do you have an idea what is going on here?

sfuhrer, 2021-01-18:
still valid @dlwalter ?

dlwalter, 2021-01-18:
I'd be ok with closing it, I won't be able to test this for a couple days.

closed 2021-01-18 by sfuhrer