PX4/PX4-Autopilot #12955: Listener wrongly showing "never published"

ItsTimmy, 2019-09-13:
**Describe the bug**
When running the "listener" command on the PX4 shell, it sometimes shows "never published" for topics that have absolutely been published. This happens when using the `-n` or `-r` arguments and NOT `-i`. Below are some examples (with the topic data removed, for brevity):

```
nsh> listener vehicle_attitude
  [correct topic data]
nsh> listener vehicle_attitude -i 0
  [correct topic data]
nsh> listener vehicle_attitude -i 0 -n 2
  [correct topic data twice]
nsh> listener vehicle_attitude -n 2
never published
nsh> listener vehicle_attitude -n 2 -r 2
never published
```

This behavior is the same in SITL.

**To Reproduce**
Steps to reproduce the behavior:
1. Choose a topic that has been published (Run `uorb top` to find some such topics)
2. Run the command `listener <topic>` and `listener <topic> -i 0 -n 10` and observe correct behavior
3. Run the command `listener <topic> -n 10` and observe `never published`.

**Drone (please complete the following information):**
- FMU V5 Test bench running the latest master branch as of the time of submitting this issue
- Same issue in SITL

ItsTimmy, 2019-09-13:
Perhaps related to #12799

dagar, 2019-09-13:
Yes I was aware this slipped in with https://github.com/PX4/Firmware/pull/12799 and was planning to go back and fix it.

PR #13006 (2019-09-25): listener fix incorrect "never published" cases
 - fixes #12955
commit b6db8723 (2019-09-25): listener fix incorrect "never published" cases (#13006)

- fixes #12955
closed 2019-09-25 by julianoes