PX4/PX4-Autopilot #12958: pixhawk 4 mini px4_fmu-v5 Baro Sensors 0 Missing

yasdum, 2019-09-14:
**Describe the bug**
pixhawk 4 mini px4_fmu-v5 Baro Sensors 0 Missing error

dagar, 2019-09-17:
This should already be fixed in master. I'll get Beta 2 out shortly.

closed 2019-09-30 by dagar

majid55-pixmini, 2020-01-23:
hey if anyone have any solution for this issue i am new and have no idea please help me out

julianoes, 2020-01-24:
@majid55-pixmini please create a new issue and properly describe your problem with all the output, version information  etc.

majid55-pixmini, 2020-01-25:
hey julianose.
i have pixhwak 4 mini fc and recently it is giving me these errors.
1-attitude estimator not running - check system boo
2-Pref-fight checks failed Baro Sensors 0 Missing
board have latest px4_fmu-v5  V1.10.1 stable version. updated through QGC
and then i tried to reinstall previous versions too but error doesn't go away.
i did not crash my drone and it was working good before installing latest version.
if i disable ekf2 in perimeters.
and press the safty switch on GPS i am able to arm and fly the drone but it is not holding altitude anymore and if i flip the switch for alt hold or GPS hold it say "Reject altitude Hold " Reject GPS hold "
Alt hold and GPS lock was working before only new hardware i added is telemetry device and updated to latest V1.10.1.
so what could be the reason i am sure it is software reason because when i google this error lot's of people have this issue after updating the software so is there any fix for this issue.
Thanks i advance

julianoes, 2020-02-12:
@majid55-pixmini as written above, please create a new issue and add a log.