PX4/PX4-Autopilot #12963: make px4_sitl jmavsim    fail

onionysy, 2019-09-16:
We use GitHub issues only to discuss PX4 bugs and new features. For
questions about using PX4 or related components, please use [PX4 Discuss](http://discuss.px4.io/).

when i ran    make px4_sitl jmavsim 
`[0/1] cd /home/coamc/Firmware/build/px.../coamc/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/coamc/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /home/coamc/Firmware
build_path: /home/coamc/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/coamc/Firmware/build/px4_sitl_default/bin/px4" "/home/coamc/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/coamc/Firmware"/test_data
INFO  [px4] Creating symlink /home/coamc/Firmware/ROMFS/px4fmu_common -> /home/coamc/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  CAL_BARO_PRIME: curr: 0 -> new: 6620172
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.1000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_JERK_MIN: curr: 8.0000 -> new: 10.0000
  MPC_ACC_HOR_MAX: curr: 5.0000 -> new: 3.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
[... 128 lines cut ...]
[3/61] Building CXX object CMakeFiles/physics_msgs.dir/Wind.pb.cc.o
[4/61] Building CXX object CMakeFiles/std_msgs.dir/Int32.pb.cc.o
[5/61] Building CXX object CMakeFiles/sensor_msgs.dir/MagneticField.pb.cc.o
[6/61] Building CXX object CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o
[7/61] Building CXX object CMakeFiles/gazebo_magnetometer_plugin.dir/src/geo_mag_declination.cpp.o
[8/61] Building CXX object CMakeFiles/gazebo_motor_model.dir/src/gazebo_motor_model.cpp.o
[9/61] Building CXX object CMakeFiles/gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o
[10/61] Building CXX object CMakeFiles/gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o
[11/61] Building CXX object CMakeFiles/gazebo_sonar_plugin.dir/src/gazebo_sonar_plugin.cpp.o
[12/61] Building CXX object CMakeFiles/gazebo_barometer_plugin.dir/src/gazebo_barometer_plugin.cpp.o
[13/61] Building CXX object CMakeFiles/gazebo_wind_plugin.dir/src/gazebo_wind_plugin.cpp.o
[14/61] Building CXX object CMakeFiles/LiftDragPlugin.dir/src/liftdrag_plugin/liftdrag_plugin.cpp.o
[15/61] Building CXX object CMakeFiles/gazebo_magnetometer_plugin.dir/src/gazebo_magnetometer_plugin.cpp.o
[16/61] Building CXX object CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o
[17/61] Building CXX object CMakeFiles/gazebo_multirotor_base_plugin.dir/src/gazebo_multirotor_base_plugin.cpp.o
[18/61] Building CXX object CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:200: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
`

I want to ask what went wrong.

Thanks!

onionysy, 2019-09-16:
i use ubuntu 18.04/ros melodic

julianoes, 2019-09-16:
For jMAVSim:
Check this https://dev.px4.io/master/en/simulation/jmavsim.html#troubleshooting

For Gazebo:

```
gz: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
```

It looks like a ignition dependency mismatch. According to [this](https://benshihnotes.blogspot.com/2019/03/gazebo-in-terminal-doesnt-run-properly.html?showComment=1553783471100#c4266156222530806435), try:
```
sudo apt upgrade libignition-math2
```

julianoes, 2019-09-16:
Have you done these steps?

```
sudo apt install openjdk-8-jdk
sudo update-alternatives --config java # choose 8
rm -rf Tools/jMAVSim/out
```

onionysy, 2019-09-17:
> Have you done these steps?
> 
> ```
> sudo apt install openjdk-8-jdk
> sudo update-alternatives --config java # choose 8
> rm -rf Tools/jMAVSim/out
> ```

yes, but when i use java8, it can,t be used until i change to java11
when i use java8. it shows:
`Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 54.0), this version of the Java Runtime only recognizes class file versions up to 52.0`

julianoes, 2019-09-17:
Have you done the step `rm -rf Tools/jMAVSim/out` as well?

haslinghuis, 2019-12-04:
After setup build environment (https://github.com/PX4/Devguide/blob/master/en/setup/dev_env_linux_ubuntu.md)

Having applied
`sudo apt install openjdk-8-jdk
sudo update-alternatives --config java # choose Java 8 (select 2 in my case)
rm -rf Tools/jMAVSim/out`

trying to `$ make px4_sitl jmavsim` I get

0 WARNING: setRealTimeSched failed (not run as root?)
BUILD SUCCESFUL
Total time: 5 seconds
Options parsed, starting Sim.
Starting GUI...
Exception in thread "main" java.lang.reflect.InvocationTargetException
Caused by: java.awt.AWTERROR: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper

Quoting the fix for this error:

    This can be done by editing the accessibility.properties file for OpenJDK:
    sudo vim /etc/java-8-openjdk/accessibility.properties
    Comment out the following line:
    assistive_technologies=org.GNOME.Accessibility.AtkWrapper

This where the steps to get jMAVSIM running

julianoes, 2019-12-06:
Thanks, it's also mentioned here:
https://dev.px4.io/master/en/simulation/jmavsim.html#javaawtawterror-assistive-technology-not-found-orggnomeaccessibilityatkwrapper

closed 2019-12-06 by julianoes

mwshafer, 2022-05-26:
@julianoes Sorry for all the questions from me and drudging up a closed issue. I have gotten all of the dependencies worked out but I am now stuck with the issue discussed above. I already ran `brew tap AdoptOpenJDK/openjdk` and `brew install --cask adoptopenjdk15` per the troubleshooting [here](https://docs.px4.io/master/en/simulation/jmavsim.html). I also ran the `rm -rf Tools/jMAVSim/out` command you suggested above to no avail. My full terminal output is below. I don't know if it matters, but a day or so ago the tutorial on setting up the dev environement changed at the java step. It had been `brew install --cask temurin` but recently changed to `brew tap AdoptOpenJDK/openjdk 
       brew install --cask adoptopenjdk15`. 

mshafer@Michaels-MacBook-Pro PX4-Autopilot % make px4_sitl jmavsim         
[0/1] cd /Users/mshafer/Desktop/Desktop/PLAYGROU...UND/PX4_Sim/PX4-Autopilot/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /Users/mshafer/Desktop/Desktop/PLAYGROUND/PX4_Sim/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /Users/mshafer/Desktop/Desktop/PLAYGROUND/PX4_Sim/PX4-Autopilot
build_path: /Users/mshafer/Desktop/Desktop/PLAYGROUND/PX4_Sim/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/Users/mshafer/Desktop/Desktop/PLAYGROUND/PX4_Sim/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/Users/mshafer/Desktop/Desktop/PLAYGROUND/PX4_Sim/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/mshafer/Desktop/Desktop/PLAYGROUND/PX4_Sim/PX4-Autopilot"/test_data
Creating symlink /Users/mshafer/Desktop/Desktop/PLAYGROUND/PX4_Sim/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/m

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
Buildfile: /Users/mshafer/Desktop/Desktop/PLAYGROUND/PX4_Sim/PX4-Autopilot/Tools/jMAVSim/build.xml
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 62.0), this version of the Java Runtime only recognizes class file versions up to 59.0
	at java.base/java.lang.ClassLoader.defineClass1(Native Method)
	at java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1016)
	at java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:151)
	at java.base/java.net.URLClassLoader.defineClass(URLClassLoader.java:514)
	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:422)
	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:416)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:691)
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:415)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:468)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:59)