PX4/PX4-Autopilot #12971: MC stabilized/altitude control setpoint shaping

bresch, 2019-09-16:
### Context
In stabilized/altitude control modes, the sticks of the remote are directly mapped to attitude setpoints. This means that the drone tries to follow the dynamic imposed by the pilot, which is sometimes not desirable. Indeed, a lot of users prefer to have "smooth" stick response rather than tight tracking and to achieve this result, they are using one (or a combination) of the following tricks:
1. lower the saturation of the rate controllers (i.e.: [MC_ROLLRATE_MAX](https://dev.px4.io/master/en/advanced/parameter_reference.html#MC_ROLLRATE_MAX), ...)
2. de-tune the rate controller (lower the rate controller gains)
3. reduce the attitude proportional gain (i.e.: [MC_ROLL_P](https://dev.px4.io/master/en/advanced/parameter_reference.html#MC_ROLL_P))

However, those parameters are not meant to be tune for the "stick feel" but for the actual stability of the drone. It means that if lowering those parameters is not required from a stability/robustness point-of-view, this creates bad side effects. The drone might have bad disturbance rejection or even become unstable when a large disturbance is applied (especially when the input of the rate controller is highly saturated). 
Furthermore, in cascaded control, the tuning of the inner loop is a hard limiting factor of the performance of the outer one: modifying the tuning of the rate controller has an impact on the attitude, velocity and position loops.

_NASA didn't de-tune the Grumman Gulfstream II to fly like the space shuttle, the flight controller **simulates** its handling qualities._

To summarize: one should **NEVER** modify the tuning of a control loop to adjust the stick feel. The problem is that we currently don't have anything to tune the stick feel and we need to add that.

### Possible solutions
I currently see two possible solutions with their pros and cons. We could implement both to test and compare or to give the choice to the user. Maybe someone has other ideas as well.
#### a. Slew-rate
Apply a slew-rate on the RC input (or manual attitude setpoint): this will give the same feeling as limiting `MC_ROLLRATE_MAX` without affecting the stability/disturbance rejection.

_Pro_:The maximum rate is defined by this parameter and doesn't depend on the amplitude of the stick input
_Con_: The drone doesn't act as a linear system as the dynamic is constrained; if this is too much constrained, the pilot might feel a lag on the response.

#### b. Dynamic filter
Apply a first order filter on the RC input (or manual attitude setpoint): this simulates the effect of having a lower attitude proportional gain without affecting the stability/disturbance rejection.

_Pro_: The drone will still act as a dynamic system, this filter can be tuned such that the vehicle behaves like a heavier one (a racer can react like a 40kg drone if required).
_Con_: The maximum rate is not directly constrained, the commanded angular rate depends on the amplitude of the commanded motion (nothing unsafe here, it can't exceed the input saturation of the rate controller).

### Additional changes
I would like to rename the `_MAX` parameters that are used to saturate the controllers with the suffix `_SAT` such that the users are not tempted to use those to change the attitude response.  Maybe becomes useless if we can have a page in QGC for those tuning parameters where the description is clear and the name of the parameter isn't relevant. If the parameter change isn't something we want to do, we can still improve the description of the parameter in order to warn the user.

FYI @MaEtUgR

Stifael, 2019-09-17:
The customers that you have in mind, what are they trying to accomplish with the altitude mode?

bresch, 2019-09-24:
@Stifael Safety fallback, FPV filming, ... The usual complain is that the drone feels too aggressive.
I think I'll give a try with a first order filter; it's quick to implement and testing is the only way to see if the feeling is the desired one.

Stifael, 2019-09-24:
because Altitude is also the safety fallback for position control, I would rather create a new FlightTask that implements the requested behavior. For instance, a solution could be FlightTaskManualPosition but without position lock

MaEtUgR, 2019-09-28:
> I would like to rename the _MAX parameters that are used to saturate the controllers with the suffix _SAT

I would propose `_LIM` for limit, just an opinion because saturate is a word that's probably less used and sat reminds me of satellite.

For the simulation of the stick feel may I suggest to have a mode just like Altitude (keepig the original one available like @Stifael suggests) but instead of the right stick directly commanding the orientation of the vehicle it sets an acceleration setpoint with jerk limit? The advantage I see with this approach is the user can tune accelerations and jerks for setpoint shaping once and it will feel the same in position and altitude mode just that in position the vehicle brakes automatically and doesn't drift afterward since it has also velocity and position estimate available.

closed 2020-05-20 by MaEtUgR