PX4/PX4-Autopilot #12974: FMUv4: CPU load > 95%

nicovanduijn, 2019-09-17:
**Describe the bug**
With current master, I'm seeing > 96% CPU load on FMUv4 (pixracer), leading to log dropouts etc.

It seems that the load is high on boot, and then drops to around ~80% after a few seconds, and increases back to 95% when arming.

Tested with 270e12a4dfbe5ea24035483beabca90f3bd4b18d

I have not yet found the time to check other FMUs.

**To Reproduce**
Just flash a pixracer and arm it.

**Log File**
https://review.px4.io/plot_app?log=695a3060-e155-4622-b625-d5b870a50871

dagar, 2019-09-17:
* mavlink (2 instances) ~= 17% cpu
       * Can you run `mavlink status streams` on a full operational system? I would guess we this could be trimmed.
       * https://github.com/PX4/Firmware/pull/12967 might save a small amount
  * wq:rate_ctrl (inner loop + output module) ~= 11%
       * This can be decreased slightly by setting `IMU_GYRO_RATEMAX` to 400 (your PWM_RATE)

nicovanduijn, 2019-09-17:
Here's the output of `mavlink status streams`: (Sorry for the screenshot, QGC doesn't let me copy-paste)

[image]
[image]

dagar, 2019-09-17:
[image]

dagar, 2019-09-30:
Should be resolved, although there's plenty of room for further improvement.

closed 2019-09-30 by dagar