PX4/PX4-Autopilot #12979: Printouts to console only after module restart

JMshare, 2019-09-17:
**Describe the bug**
For some reason, the printouts to console (issued by `PX4_WARN`, `PX4_INFO` or `PX4_ERR`) don't show if the module is started from the airframe startup file. If I want to see the printouts I have to manually stop the module and start it again through the console.

**To Reproduce**
Create a custom module printing something out.
Put `my_module start` to the airframe startup file.
See no printouts.
Stop and start the module through the Mavlink console in QGC.
See printouts.

**Expected behavior**
I suppose one puts the printouts to the code for reason, and having to restart the module annihilates the benefits of starting it automatically through the startup script. Moreover, if you realise you forgot to restart it once you are already in the air, it may be too late. 

**Drone info**
px4_fmu-v5, 1.9.0 fw

**Additional context**
If I remember right this feature was added somewhere along the 1.8 and 1.9 transition.

dagar, 2019-09-19:
If you give this a try with master (soon to be v1.10.0) you'll be able to get this output via `dmesg`.

julianoes, 2019-09-20:
To get the printfs you need to use either a debug proble (see https://dev.px4.io/master/en/debug/system_console.html) or use dmesg. I don't think this is easily doable otherwise.

closed 2019-09-20 by julianoes