PX4/PX4-Autopilot #12980: Yawrate in Offboard Stopped Working

Pedro-Roque, 2019-09-17:
**Describe the bug**
Yawrate commands sent through mavros/setpoint_raw/attitude as bodyrates in Z, with appropriate mask, stopped working. 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on
2. Publish a command on mavros/setpoint_raw/attitude with a yawrate and correct mask
3. The vehicle does not respond to yawrate commands

**Expected behavior**
The vehicle should rotate according to the bodyrate Z command

**Log Files and Screenshots**
Here goes a folder with 2 rosbags and the flight logs:
https://drive.google.com/open?id=1jufiucPZ2O7UPZHpCZfKG8gnDEj1XmIC

**Drone (please complete the following information):**
- Hover 1 Frame
- Pixhawk Radiolink, with v3 bootloader and v1.9.2 stable

**Additional context**
Recorded on a MoCap enabled arena at KTH Smart Mobility Lab.

Pedro-Roque, 2019-09-17:
@Jaeyoung-Lim @mhkabir @MaEtUgR

Pedro-Roque, 2019-09-18:
Update: in SITL, with IRIS, it checks out fine, that is, the system responds as expected! 

I am using the Generic Quadrotor X frame. Something I should be careful with? Maybe I should change the yaw gain?

Pedro-Roque, 2019-09-18:
Update: Flying with master does not show the issue! Just successfully flied with mas flashed without any problems. Maybe there was something missing on the v1.9.2 stable...

Anyhow, I will follow @dagar 's suggestion and add a test for this next week - hopefully before next release.

MaEtUgR, 2019-09-19:
> tried to test SITL, but get compass #0 uncalibrated immediately once I try to arm the vehicle.
Do a make clean if that happens with SITL, it's an incompatibility of the configuration that requires rebuilding after switching the branch across a certain point.

Pedro-Roque, 2019-09-19:
Will do, next week though! :)

Jaeyoung-Lim, 2019-09-25:
@Pedro-Roque Any updates?

Pedro-Roque, 2019-09-30:
@Jaeyoung-Lim not yet but not forgotten. Just need to finish some stuff on my research and then will look into this.

julianoes, 2020-01-30:
Bump :smile:

Pedro-Roque, 2020-02-03:
Not forgotten @julianoes ! I'm just trying to dodge deadlines these days, but I'll come back to this as soon as I can

Pedro-Roque, 2020-05-03:
Issue fixed and tests added. Closing

closed 2020-05-03 by Pedro-Roque