PX4/PX4-Autopilot #12987: CM8JL65 switches uORB instance mid-air

nicovanduijn, 2019-09-19:
**Describe the bug**
I've seen a bug where the distance sensor CM8JL65 switches the uORB topic it is publishing on, mid-air. This doesn't seem to bother EKF2 much, but it still doesn't sound like a great idea.

This was seen when testing a rebased version of https://github.com/PX4/Firmware/pull/12950, but I don't think the issue is related to the PR.

Would be interesting to know if anybody has seen similar behavior with other sensors, or if this is more likely a bug in the CM8JL65 driver.

**Screenshots**
[image]

**Logs**
[Log file](https://review.px4.io/plot_app?log=c5c192d0-ae47-4bbc-a8de-2c887bddaf87)

kamilritz, 2019-09-19:
I have experienced the same:

**Screenshot**
[image]
 **Logs**
[Log file](https://review.px4.io/plot_app?log=bd994025-2312-4ee9-a7b2-f6ab2d65c6b8)

dagar, 2019-09-19:
Could you try https://github.com/PX4/Firmware/pull/12993?

dagar, 2019-09-21:
If I look in Flightplot or dump the data to csv I see a 2nd instance of distance_sensor, but it only publishes 2 data points. Is this a plotjuggler bug?

nicovanduijn, 2019-09-24:
This is weird, if I use FlightPlot instead of plotjuggler, I see something slightly different.
PlotJuggler:
- 3 Instances of distance sensor: `distance_sensor`, `distance_sensor_0` and `distance_sensor_1`. With instance 1 holding just two samples as you described.
[image]

- In flightPlot it seems that the two instances `distance_sensor` and `distance_sensor_0` get fused. There is no `distance_sensor` anymore, but the weird instance with just two samples still exists:
[image]

dagar, 2019-09-30:
Can you try again with master?

julianoes, 2020-01-30:
No answer given, closing.

closed 2020-01-30 by julianoes