PX4/PX4-Autopilot #12990: Rover - Aion Robotics R1 UGV issues with executing position setpoints

sarahToonsi, 2019-09-19:
Hi, 
I was trying to send position waypoints to a differential steering vehicle with the PX4 frame set us : 
  Aion Robotics R1 UGV 
I am working with a Pixhawk 2 cube and publishing waypoints through mavros to the topic: mavros/setpoint_position/local
My vehicle starts to move but it never steers even when the waypoints is to its right or left. It keeps going forward or backward with full throttle and I don't have a clear description to its random movement. I'm working in the lab with a Mocap system and my position settings are fine. 
The parameters that  I changed in Qground control are only for the EKF to use vision instead of GPS and for setting up the companion computer.

julianoes, 2019-09-19:
Could you please upload a log to logs.px4.io and link it here? Thanks.

sarahToonsi, 2019-09-24:
>  logs.px4.io

https://logs.px4.io/plot_app?log=04064ce8-7e43-4e3d-af40-d6c098599da2

julianoes, 2019-09-24:
FYI @ItsTimmy @Jaeyoung-Lim

Jaeyoung-Lim, 2019-09-24:
@sarahToonsi The rover position controller requires you to have a GPS fix.

LorenzMeier, 2019-09-30:
Original issue captures this already. This is not a real issue but rather a requirement.

closed 2019-09-30 by LorenzMeier