PX4/PX4-Autopilot #12996: Crazyflie 2.0 how to config "make bitcraze_crazyflie_default qconfig"

waytry, 2019-09-20:
**Describe the bug**
[Crazyflie 2.0 · PX4 User Guide](https://docs.px4.io/master/en/complete_vehicles/crazyflie2.html#flashing-px4) has the following

> If QGroundControl does not connect with the vehicle, ensure that in nuttx-config for crazyflie # CONFIG_DEV_LOWCONSOLE is not set is replaced by CONFIG_DEV_LOWCONSOLE=y. This should be done using menuconfig:
> 
>   make bitcraze_crazyflie_default menuconfig
> or qconfig (Check Low-level console support under Serial Driver Support in GUI):
> 
>   make bitcraze_crazyflie_default qconfig

Everything worked fine and "make bitcraze_crazyflie_default upload" was successful initially. But after I _Check Low-level console support under Serial Driver Support in GUI_ and "make bitcraze_crazyflie_default upload" again, I received the following error message:

> pan@pan-VB:~/src/Firmware$ make bitcraze_crazyflie_default upload
> -- PX4 version: v1.10.0-beta1-87-g78bf12f0db
> -- PX4 config file: /home/pan/src/Firmware/boards/bitcraze/crazyflie/default.cmake
> -- PX4 config: bitcraze_crazyflie_default
> -- PX4 platform: nuttx
> -- PX4 lockstep: disabled
> -- cmake build type: MinSizeRel
> -- The CXX compiler identification is GNU 7.2.1
> -- The C compiler identification is GNU 7.2.1
> -- The ASM compiler identification is GNU
> -- Found assembler: /home/pan/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
> -- Check for working CXX compiler: /home/pan/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
> -- Check for working CXX compiler: /home/pan/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Check for working C compiler: /home/pan/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
> -- Check for working C compiler: /home/pan/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
> -- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
> -- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-30-g3b32ee4
> -- ROMFS: px4fmu_common
> -- ROMFS: Adding rc.board_defaults
> -- ROMFS: Adding rc.board_sensors
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/pan/src/Firmware/build/bitcraze_crazyflie_default
> [0/689] git submodule platforms/nuttx/NuttX/nuttx
> [1/689] git submodule platforms/nuttx/NuttX/apps
> [2/689] git submodule src/lib/ecl
> [7/689] git submodule src/drivers/gps/devices
> [8/689] git submodule mavlink/include/mavlink/v2.0
> [686/689] Linking CXX executable bitcraze_crazyflie_default.elf
> FAILED: bitcraze_crazyflie_default.elf 
> : && /home/pan/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++  -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -DNDEBUG  --specs=nosys.specs platforms/nuttx/CMakeFiles/px4.dir/__/common/empty.c.obj  -o bitcraze_crazyflie_default.elf -L/home/pan/src/Firmware/build/bitcraze_crazyflie_default/external/Install/lib -nostartfiles -nodefaultlibs -nostdlib -nostdinc++ -fno-exceptions -fno-rtti -Wl,--script=/home/pan/src/Firmware/build/bitcraze_crazyflie_default/NuttX/nuttx-config/scripts/script.ld -Wl,-Map=bitcraze_crazyflie_default.map -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group src/drivers/boards/common/stm32/libdrivers_boards_common_arch.a NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/libs/libc/libc.a NuttX/nuttx/configs/libconfigs.a NuttX/nuttx/libs/libxx/libxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -Wl,--end-group -lm -lgcc src/modules/uORB/libmodules__uORB.a boards/bitcraze/crazyflie/syslink/libmodules__syslink.a src/drivers/barometer/lps25h/libdrivers__lps25h.a src/drivers/distance_sensor/vl53lxx/libdrivers__vl53lxx.a src/drivers/gps/libdrivers__gps.a src/drivers/imu/mpu9250/libdrivers__mpu9250.a src/drivers/optical_flow/pmw3901/libdrivers__optical_flow__pmw3901.a src/drivers/px4fmu/libdrivers__px4fmu.a src/modules/attitude_estimator_q/libmodules__attitude_estimator_q.a src/modules/camera_feedback/libmodules__camera_feedback.a src/modules/commander/libmodules__commander.a src/modules/dataman/libmodules__dataman.a src/modules/ekf2/libmodules__ekf2.a src/modules/events/libmodules__events.a src/modules/land_detector/libmodules__land_detector.a src/modules/landing_target_estimator/libmodules__landing_target_estimator.a src/modules/load_mon/libmodules__load_mon.a src/modules/local_position_estimator/libmodules__local_position_estimator.a src/modules/logger/libmodules__logger.a src/modules/mavlink/libmodules__mavlink.a src/modules/mc_att_control/libmodules__mc_att_control.a src/modules/mc_pos_control/libmodules__mc_pos_control.a src/modules/navigator/libmodules__navigator.a src/modules/sensors/libmodules__sensors.a src/modules/sih/libmodules__sih.a src/modules/airspeed_selector/libmodules__airspeed_selector.a src/systemcmds/bl_update/libsystemcmds__bl_update.a src/systemcmds/config/libsystemcmds__config.a src/systemcmds/dumpfile/libsystemcmds__dumpfile.a src/systemcmds/esc_calib/libsystemcmds__esc_calib.a src/systemcmds/hardfault_log/libsystemcmds__hardfault_log.a src/systemcmds/i2cdetect/libsystemcmds__i2cdetect.a src/systemcmds/led_control/libsystemcmds__led_control.a src/systemcmds/mixer/libsystemcmds__mixer.a src/systemcmds/motor_ramp/libsystemcmds__motor_ramp.a src/systemcmds/motor_test/libsystemcmds__motor_test.a src/systemcmds/mtd/libsystemcmds__mtd.a src/systemcmds/nshterm/libsystemcmds__nshterm.a src/systemcmds/param/libsystemcmds__param.a src/systemcmds/perf/libsystemcmds__perf.a src/systemcmds/pwm/libsystemcmds__pwm.a src/systemcmds/reboot/libsystemcmds__reboot.a src/systemcmds/reflect/libsystemcmds__reflect.a src/systemcmds/sd_bench/libsystemcmds__sd_bench.a src/systemcmds/shutdown/libsystemcmds__shutdown.a src/systemcmds/top/libsystemcmds__top.a src/systemcmds/topic_listener/libsystemcmds__topic_listener.a src/systemcmds/tune_control/libsystemcmds__tune_control.a src/systemcmds/usb_connected/libsystemcmds__usb_connected.a src/systemcmds/ver/libsystemcmds__ver.a ROMFS/libromfs.a platforms/nuttx/src/px4/stm/stm32f4/io_pins/libarch_io_pins.a src/lib/mixer_module/libmixer_module.a src/lib/output_limit/liboutput_limit.a src/lib/DriverFramework/framework/src/libdf_driver_framework.a src/modules/commander/failure_detector/libfailure_detector.a src/lib/ecl/EKF/libecl_EKF.a src/lib/ecl/geo_lookup/libecl_geo_lookup.a src/lib/circuit_breaker/libcircuit_breaker.a src/modules/mc_att_control/AttitudeControl/libAttitudeControl.a src/lib/controllib/libcontrollib.a src/lib/FlightTasks/libFlightTasks.a src/lib/FlightTasks/tasks/ManualAltitudeSmoothVel/libFlightTaskManualAltitudeSmoothVel.a src/lib/FlightTasks/tasks/ManualPositionSmooth/libFlightTaskManualPositionSmooth.a src/lib/FlightTasks/tasks/ManualPositionSmoothVel/libFlightTaskManualPositionSmoothVel.a src/lib/FlightTasks/tasks/ManualPosition/libFlightTaskManualPosition.a src/lib/FlightTasks/tasks/AutoLine/libFlightTaskAutoLine.a src/lib/FlightTasks/tasks/AutoMapper/libFlightTaskAutoMapper.a src/lib/FlightTasks/tasks/AutoLineSmoothVel/libFlightTaskAutoLineSmoothVel.a src/lib/FlightTasks/tasks/AutoMapper2/libFlightTaskAutoMapper2.a src/lib/FlightTasks/tasks/AutoFollowMe/libFlightTaskAutoFollowMe.a src/lib/FlightTasks/tasks/Auto/libFlightTaskAuto.a src/lib/FlightTasks/tasks/Offboard/libFlightTaskOffboard.a src/lib/FlightTasks/tasks/Failsafe/libFlightTaskFailsafe.a src/lib/FlightTasks/tasks/Transition/libFlightTaskTransition.a src/lib/FlightTasks/tasks/Orbit/libFlightTaskOrbit.a src/lib/FlightTasks/tasks/ManualAltitudeSmooth/libFlightTaskManualAltitudeSmooth.a src/lib/FlightTasks/tasks/ManualAltitude/libFlightTaskManualAltitude.a src/lib/FlightTasks/tasks/Manual/libFlightTaskManual.a src/lib/FlightTasks/tasks/Utility/libFlightTaskUtility.a src/lib/hysteresis/libhysteresis.a src/lib/FlightTasks/tasks/FlightTask/libFlightTask.a src/lib/WeatherVane/libWeatherVane.a src/lib/CollisionPrevention/libCollisionPrevention.a src/modules/mc_pos_control/Takeoff/libTakeoff.a src/lib/ecl/geo/libecl_geo.a src/lib/landing_slope/liblanding_slope.a src/lib/airspeed/libairspeed.a src/lib/conversion/libc [... cut ...]
> Makefile:193: recipe for target 'bitcraze_crazyflie_default' failed
> make: *** [bitcraze_crazyflie_default] Error 1

It seems like it's something related to the _undefined reference to `stm32_serial_dma_poll'_, can anyone help me with this? Thanks!

**To Reproduce**
platform: Ubuntu 18.04

```
git clone https://github.com/PX4/Firmware.git

make bitcraze_crazyflie_default qconfig
```
Check Low-level console support under Serial Driver Support in GUI
```
make bitcraze_crazyflie_default upload
```

jfm92, 2019-09-20:
Hi,
I have the same issue, I think that is a problem with the compression of the NSH defconfig.
I partially solve the problem doing the next steps:
- Download PX4 from this commit:
https://github.com/PX4/Firmware/commit/edd9f91a190c85338cf4ef3063b2457d2347562f#diff-7fe3cafd1c025d85a850ea2cec8bf057

-  Add ``CONFIG_DEV_LOWCONSOLE=y `` without using menuconfig or qconfig, just adding to the text. (This is not correct, but it works)

This solve the communication problem with QGC, but is not a solution.

After checking the code of NuttX, looking for ``stm32_serial_dma_poll`` function, I've found that depends of the flag ``CONFIG_STM32_HCIUART_RXDMA``. But is always as no.

julianoes, 2019-09-20:
Any idea @davids5

waytry, 2019-09-20:
@jfm92 Thanks for the reply. I followed your instructions and it worked, although it's a workaround. For those of who are interested, I I listed what I did at the end.

Steps:
git clone https://github.com/PX4/Firmware.git

jfm92, 2019-09-23:
Hi @waytry,

About the calibration, I have the Crazyflie 2.1 and it has a different gyro and barometer, so at this moment I can't do the calibration. But I suspect that also is not doing a proper initialization of the peripherals.
About the wireless communication, I'm not sure what I did but I succeed once. I'll try to replicate it.
I was doing more test to figure out was going on, and if you copy the exact defconfig file, it works until the version 1.9.2 of PX4 (I mean the QGC connection).

waytry, 2019-09-23:
Hi @jfm92,

I realize that I am also working on a Crazyflie 2.1, rather than Crazyflie 2.0. As you mentioned, the version 2.1 has different IMU and barometer, which I believe causes the calibration issue. How did you handle installing PX4 on a Crazyflie 2.1? Is the version 2.1 supported now (from what I've seen only version 2.0 is supported at this moment). Did you develop drivers for the new IMU and barometer used on Crazyflie 2.1?

jfm92, 2019-09-25:
Hi @waytry 

I think that I almost solved the problem with QGC communication.
You only need to execute the menuconfig:
``make bitcraze_crazyflie_default menuconfig``

Go to: Devices Drivers -> Serial Driver Support -> Serial Console . And select the USART3.
This works for my, with the USB connection.

About the sensors, I don't developed any new driver. I found that on the version 1.10 beta1 they add the BMI088 driver which is the IMU that embedded this version of the drone. But the barometer is not developed yet.

dagar, 2019-09-30:
How different is the 2.1? Do we need a new board config or try to support both in place?

dagar, 2019-09-30:
This might help slightly. https://github.com/PX4/Firmware/pull/13056
Once connected with the QGC console run `dmesg`.

shrit, 2020-04-04:
Keep open

dagar, 2020-07-27:
Should be fixed in https://github.com/PX4/Firmware/pull/15421.

closed 2025-02-14 by mrpollo