PX4/PX4-Autopilot #12999: [Release 1.9] Multicopter loose height on full throttle in ALTCTL and POSCTL mode

svpcom, 2019-09-20:
**Describe the bug**
Multicopter loose height on full throttle in ALTCTL and POSCTL. If throttle reduces it returns on desired height. No height lost in MISSION mode or on slow flight. 

**To Reproduce**
Steps to reproduce the behavior:
1. Fly in POSCTL or ALTCTL
2. Forward with full throttle
3. See 10m height loss on 150m distance.

**Expected behavior**
Significant height loss is not expected. No height loss with 1.8 release.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=076d0591-06a2-4ae4-907e-4357acd484bc
See at 2:08

**Drone (please complete the following information):**
- hexacopter (S550 frame) with pixracer (FMUv4)
- baro is shielded using foam
- fmu and companion computer are shielded from wind using external case

bresch, 2019-09-20:
Hi @svpcom , thanks for reporting.

The behavior between 1.8 and 1.9 should be the same in altitude mode given that the implementation didn't change; my best guess is that the tilt parameter is different or that your vehicle is different. Do you have a log of the same vehicle in v1.8?

From the low you show, I see that the vertical thrust saturates and the vehicle goes down because it cannot provide more power to keep the altitude at this tilt angle.
The reason why you don't see that in position and auto modes is that they have axis prioritization (something that the altitude mode does not have). @MaEtUgR or @Stifael  Do you remember if altitude mode used to have prioritization?

I quickly tried in SITL in 1.8.2 and master and I'm always able to make the drone loose altitude by flying at high tilt angle' I don't see a regression.

Stifael, 2019-09-20:
> Do you remember if altitude mode used to have prioritization?
No, it does not prioritize. The tilt is the only parameter that saturates the thrust-sp component in xy. 

It wouldn't be too difficult to prioritize the altitude as well though

Stifael, 2019-09-20:
> Multicopter lose height on full throttle in ALTCTL and POSCTL. If throttle reduces it returns on desired height.

To me that sounds more like it is due estimation and is a common problem caused by the barometer

svpcom, 2019-09-20:
@Stifael  gps and baro heights are consistent in telemetry data. It is not a barometer issue

svpcom, 2019-09-20:
@Stifael I have not found a log with significant height loss in POSCTL (only height oscillations at accel / stop ). It seems this is only ALTCTL issue.
https://review.px4.io/plot_app?log=163e72b3-ad00-41b5-a87a-36fdeb13d7e4
small oscillations at 3:25

bresch, 2019-09-22:
So we would need to add Z prioritization in altitude mode if we want to fix that.

If you see a similar problem in position control (which has the prioritization), it means that your system saturates before 100% thrust, which breaks the prioritization logic.

MaEtUgR, 2019-09-26:
Now that I saw this issue I remember observing the same thing while flying in altitude mode before and was assuming it's a problem with my setup. The reason it's different from position mode is that the prioritization logic https://github.com/PX4/Firmware/blob/190c817a9e2e37c99111e249b5b23fd076ddfed3/src/modules/mc_pos_control/PositionControl.cpp#L298-L301 is not ran in altitude mode. Independent of this particular issue I changed the controller to always run the exact same path including prioritization in #12072 and hence the problem should be gone when it's in.

svpcom, 2019-12-25:
I still need it

svpcom, 2020-03-24:
I still need it

MaEtUgR, 2020-03-24:
@svpcom Thanks for the reminder, sorry for not updating this conversation!
I fixed the problem I described in https://github.com/PX4/Firmware/issues/12999#issuecomment-535426261 which I'm still assuming is the cause why you had these different flight characteristics in https://github.com/PX4/Firmware/pull/14017/. Could you please check if that solves your issue?

The axis prioritization is now exactly the same in multicopter altitude and position mode, only the setpoint generation differs.

closed 2020-06-25 by bresch