PX4/PX4-Autopilot #13001: MAVLink system boot did not complete in 20 seconds

julianoes, 2019-09-21:
**Describe the bug**
I get the following error in SITL:
`ERROR [mavlink] system boot did not complete in 20 seconds`

**To Reproduce**
`make px4_sitl jmavsim`

**Expected behavior**
This error should not appear.

@dagar I assume this has been introduced recently?

dagar, 2019-09-21:
> @dagar I assume this has been introduced recently?

Yes - https://github.com/PX4/Firmware/pull/12968

We should check the timing here to make sure it's actually working as intended (20 seconds from system start), but other than that I think 20 seconds is arbitrary. It's really only a fallback if the system doesn't hit `mavlink boot_complete` for whatever reason.

julianoes, 2019-09-23:
> It's really only a fallback if the system doesn't hit `mavlink boot_complete` for whatever reason.

I wonder why this happens for SITL now.

PR #13018 (2019-09-25): mavlink: always check stream subscriptions
**Describe problem solved by the proposed pull request**
This fixes the regression where we saw this in SITL:
```
ERROR [mavlink] system boot did not complete in 20 seconds
```

The reason was that the stream subscription requests were not checked because the while loop was not running yet because mavlink was not booted completely.
By moving the stream subscription requests into a function we can run them even if we don't run the rest of the loop.

**Test data / coverage**
Tested in SITL.

Fixes #13001.
closed 2019-09-25 by julianoes