PX4/PX4-Autopilot #13003: TF Mini Plus UART and I2C driver

gian1312, 2019-09-22:
Hi
In my opinion, the [TF Mini Plus](http://en.benewake.com/product/detail/5c345cd0e5b3a844c472329b.html) would be a handy and cheap rangefinder. It is capable of the I2C and the UART communication. Its predecessor is already implemented as serial device.  According to a Benewake [document](https://github.com/PX4/Firmware/files/3639760/Application.of.TFmini.Plus.in.PX4.pdf), this sensor (UART)  already worked on PX4. If it is still working, I am really sorry. I can't figure out how to get it working in this case. However, it would be nice if an I2C driver could be implemented. I couldn't get anything usefull written, though. 
As I was told, they have just finished the driver for Ardupilot together with Benewake. Maybe, this helps. 

An additional question. Is there any documentation for the Pixhawk 4 mini which states which of the different serial ports correspond to which physical port of the mini?

Best Regards

dagar, 2019-09-22:
It looks like a fairly straightforward driver to add if anyone has this hardware in hand.

mcsauder, 2019-09-23:
Hi @gian1312 , I have a TF Mini Plus that I've been using with the [tfmini driver](https://github.com/PX4/Firmware/tree/master/src/drivers/distance_sensor/tfmini) for months now, it works great for me.

As an example of what I just tested on the bench, plug the sensor into the UART & I2CB port (ttyS3) on a pixhawk 4 (fmu-v5), the run `tfmini start -d /dev/ttyS3`  If you are trying to run it on ttyS1 or ttyS2 you might be conflicting with a mavlink stream that would need to be disabled to run the sensor on those ports.  Feel free to find me on PX4 slack if you have trouble getting it working!

[image]

gian1312, 2019-09-23:
Thanks a lot for your help @mcsauder .  I gave it a short try because I had to leave home quite early today and I won't be back before next weekend. I will contact you on Slack as soon as I am back home and did some more research. For now, I got the following output:
[image]

However, the feature request for an I2C driver would still be active. (It would free the UART port for other applications).

dagar, 2019-09-30:
Isn't this a completely different driver?

mcsauder, 2019-09-30:
Hi @gian1312  and @dagar, yes. this would essentially be a (new) addition for i2c capability.  Only newer versions of the [product manual](https://acroname.com/sites/default/files/assets/sj-pm-tfmini_plus_a04_product_mannual_en.pdf) describe interfacing with the device via i2c, so it looks like its possible.  I'd like to try this out, but it will be a few weeks before I can get to it.

dimon777, 2020-01-29:
@mcsauder do you use specific branch for:
`tfmini start -d /dev/ttyS3`

I get different output when I try it:
`
nsh> tfmini start -d /dev/ttyS3
nsh> 
nsh> tfmini status
state @ 20024bb0
Using port '/dev/ttyS3'
tfmini_read: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
tfmini_com_err: 0 events
poll interval:  9 ticks
report queue	2/64 (2/2 @ 200247b0)
`

bys1123, 2020-08-10:
I need this driver too.

gian1312, 2020-12-25:
Are there any news?

FeCLN, 2021-01-25:
@mcsauder hellooo, have you connected in the UART 6 parameter to use it?

chahatdeep, 2021-06-12:
Was this solved? I'm not able to get my TFMINI-S working on UART but my older TFMINI (Not S model) works perfectly with RANGE FINDER TYPE = 20.